hungpham2511/toppra #2: Constant Time Trajectories

JadBatmobile, 2018-07-22:
Is it possible to plan trajectories with a specific duration?

hungpham2511, 2018-07-23:
Hi @JadBatmobile, 

Yes, it is possible to plan/retime trajectories to specific durations if one such parameterization exists. However, this feature is currently not implemented, but it will definitely be in near future. 

Stay tuned,
Hung

JadBatmobile, 2018-07-23:
Great, Can we talk about how i can help with is more? can we speak privately? jd224234@dal.ca

hungpham2511, 2018-07-23:
The idea is to compute two parameterizations, the fastest and the slowest and then interpolate them to get the parameterization with the desired duration.

The fastest parameterization can be obtained using the current implementation. The slowest one is not, but can be implemented quite easily: simply select the lowest acceleration at each step in the forward stage. Then you can consider the family of parameterizations 

alpha * fastest_param + (1-alpha) * slowest_param

and find one with the specified duration.

JadBatmobile, 2018-07-24:
so you are saying:

fastest_duration <- current implementation
slowest_duration <- current implementation + lowest accel. at each step in the forward stage.

desired_duration = alpha*fastest_duration + (1-alpha) * slowest_duration

solve for alpha?

hungpham2511, 2018-07-25:
@JadBatmobile That's correct. You can refer to this [function](https://github.com/hungpham2511/toppra/blob/master/toppra/algorithm/reachabilitybased/time_optimal_algorithm.py#L21) for the current implementation of the forward pass as a starting point.

hungpham2511, 2018-07-25:
Hmm @JadBatmobile I think I missed this line you wrote

desired_duration = alpha*fastest_duration + (1-alpha) * slowest_duration

It is actually not correct. The duration here must be computed from the parameterization, which is the vector xs := [x_0, x_1,....,x_n]. 

In any case, I will try to implement that functionality later this week. Thank you for your interest.

JadBatmobile, 2018-07-25:
I would seriously appreciate that since my application involves moving targets and it would be much easier if i can generate trajectories with fixed time.

I tried my best to do it today but the code was too much to handle wrapping my head around. If it is easy for you to do this i would appreciate it tremendously.

Cheers from Canada.

hungpham2511, 2018-07-30:
This functionality is now implemented in commit https://github.com/hungpham2511/toppra/commit/74b99fa0de972e49fdb422c5a3ba8c072338c469
Please see examples/kinematics_duration.py for more details

closed 2018-07-30 by hungpham2511

JadBatmobile, 2018-07-30:
THANK YOU SIR

hungpham2511, 2018-07-31:
You are welcome. Glad this is useful for you. If you find any bugs, please raise a separate issue.