hungpham2511/toppra #3: instance.compute_trajectory(0, 0) function takes very long

JadBatmobile, 2018-08-03:
Hello again,

I am running your kinematics_duration.py script, but unfortunately the 
jnt_traj, aux_traj = instance.compute_trajectory(0, 0) function is taking approximately 0.3 seconds to execute. This seems much longer than the original TOPP implementation, which clocked at around 0.08-0.1 seconds.

What do you think?

JadBatmobile, 2018-08-03:
It is not specific to the constant time implementation btw, the same line in kinematics.py takes 0.28 seconds

hungpham2511, 2018-08-04:
In both examples the number of discretization point is 1001, which is really high. You should try 101 points instead as shown below:
```python
    instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=np.linspace(0, 1, 101),
                           solver_wrapper='hotqpoases')
```
It should be around 5-10 times faster now. 

For an even faster speed you can try the newly implemented Seidel's solver as follow
```python
    instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=np.linspace(0, 1, 101),
                           solver_wrapper='seidel')
```
But be aware this implementation is currently quite buggy.

closed 2018-08-06 by hungpham2511