hungpham2511/toppra #4: optimized trajectory cartesian velocity with the cartesian acceleration constraints

lakshmip001, 2018-08-14:
I implemented TOPP-RA on my robot. Can I know how can I check Cartesian velocity profile after constraining Cartesian accelerations ?

lakshmip001, 2018-08-15:
I get path-position path velocity plot as shown below. But I don't see the units as shown in the figure. 
[image]. Can I know the units ? And Can I get a trapiezoidial velocity profile by constraining acceleration?

hungpham2511, 2018-08-15:
The horizontal axis is discretized path positions (variable s), while the vertical axis is square path velocity (variable sdot ^ 2). s is dimensionless, while sdot is reciprocal of second.

TOPP-RA computes *the* time-optimal parametrization/velocity profile, subject to constraints. For trapezoidal velocity profile, you might want to consult some references on robotic motion planning.

hungpham2511, 2018-08-15:
> Hi.. Thanks a lot . Can I know how to get Cartesian velocities ??

Do you mean the robot tool’s Cartesian velocity?

lakshmip001, 2018-08-15:
hi. yes after constraining the Cartesian acceleration.

hungpham2511, 2018-08-15:
If you use OpenRAVE, as in the example `catesian_accel.py`, you should look
at the function GetLinkVelocities of OpenRAVE.

You can find the relevant documentation below:
http://openrave.org/docs/latest_stable/coreapihtml/classOpenRAVE_1_1KinBody.html#a64d9e1a629412ae285a15b3767f4b3f0

hungpham2511, 2018-08-15:
In [this line](https://github.com/hungpham2511/toppra/blob/master/examples/cartesian_accel.py#L62), Cartesian acceleration limits are defined to be less than 0.5 m/s^2 in magnitude for all directions.

You are recommended to read the tutorial [here](https://hungpham2511.github.io/toppra/tutorials/2_can_linear_constraints.html), in which the constraint formulation is given.

lakshmip001, 2018-08-16:
Hi. I went through the tutorials. According to my understanding, using inequality you are just calculating the accelerations than inputing accelerations and constraining (normally done in moveit_topp). I hope my understanding is correct .

I would like to know what is qdd_full  in the following lines of code.
 qdd_full = np.zeros(robot.GetDOF())
            qdd_full[:qdd.shape[0]] = qdd

If my understanding is right you give a trajectory and later do inverse dynamics and using inequality constraint to calculate velocity and acceleration.

hungpham2511, 2018-08-16:
Hi @lakshmip001 please read [this tutorial](https://hungpham2511.github.io/toppra/tutorials/2_can_linear_constraints.html#defining-a-second-order-constraint). Everything you need to know to implement and understand how TOPP-RA handle Cartesian acceleration limit can be found in that tutorial.

q, qd, qdd are variables containing the robot joint position, velocity, and acceleration. Please try to run the example given in that tutorial.

closed 2018-08-16 by hungpham2511