hungpham2511/toppra #5: Can I use Cartesian Space path instead of Joint Space path as input for toppra？

Linjackffy, 2018-08-31:
I want to use toppra for my research. After I finish reading the documentation and tutorials, I found that the
paths are represented by Joint Space.Can I use Cartesian Space path as input for toppra?
Looking forward to your reply.

hungpham2511, 2018-08-31:
Hi @Linjackffy ,

In principle, it is possible

Suppose you want to limit the Cartesian and Velocity constraints of a point moving in 3D space. One thing you can do is to think of the point as being a 3-dof robot with translational movements along the X, Y and Z-axis respectively.

Rotational kinematic constraints will be somewhat more troublesome, but you should be able to re-formulate those constraints to the form accepted by TOPP-RA. 

Hope that help. Please let me know if you have any further question.

JadBatmobile, 2018-09-01:
Im doing something similar.. i simply translate the cartesian waypoints to joint waypoints tho, using the inverse kinematics of my model

Linjackffy, 2018-09-02:
@hungpham2511 Thanks for your reply！
Your suggestion is instructive for me. I also need to think about joint constraints like velocity, acceleration, etc, while moving along Cartisian Space path. Can I use inverse kinematics to convert the joint constraint to the cartesian constraint?

Linjackffy, 2018-09-02:
@JadBatmobile 
Your method may lose the precision of the cartesian path.

closed 2018-10-14 by hungpham2511