hungpham2511/toppra #18: about compute the reachable set

guofuyu, 2018-10-27:
I tried the function compute_controllable_set(), it works well. When I tried to implement a similar but forward function compute_reachable_set(), it seems that something was wrong. I have forked your repo, and commit my modification, as you can see on [https://github.com/guofuyu/toppra/commits/master](url)

[image]

I really appreciate that if you can help me and point where was wrong when you are free. thank for your time

hungpham2511, 2018-10-28:
@fuyu-guo  I just make some comments on your commits. Please take a look.

hungpham2511, 2018-11-09:
@fuyu-jia Please close the issue if it has been addressed.

closed 2018-11-16 by hungpham2511