hungpham2511/toppra #24: Can you give me some advice about how to implement this library in a real industrial robot

huangjiqiang, 2018-12-17:
Your library is very stable and efficient. So I want to apply this library to online trajectory planner in my robot lab. During the applying process, there is a problem that I don't know how to write the code about the real-time buffer. Computing the whole path is time-consuming. 
So Could you give me some advice about how to write a real-time buffer when applying this library?

hungpham2511, 2018-12-17:
Hi @huangjiqiang, thank you for using toppra.

Unfortunately, I can't provide any advice on the specific platform that you are using, as I am not sure how it works or what it is. 

From my own experience in our lab with Denso robots, we basically compute a trajectory and send the waypoints sequentially at a fixed cycle time. The only computational cost is from the parametrization of the geometric paths, which generally should be a few milliseconds.

huangjiqiang, 2018-12-18:
Computing a whole geometric path and then send waypoints to servo drivers in a fixed cycle time? 
Time-consuming is due to add jerk constraints to trajectory planner.

hungpham2511, 2018-12-18:
Toppra can't account for jerk constraints, so you must have done this computation elsewhere? In that case I won't be able to help.

In our lab we just send the reference positions directly w/out further processing.

hahazhky, 2018-12-22:
@hungpham2511 
I want to implement this algorithm in my real robot and need some help.
in your real robot, is position only controlled?how about velocity and acceleration? 
Thanks

hungpham2511, 2018-12-22:
Hi @hahazhky, 

In general `toppra` produces trajectories, which contain position, velocity and acceleration. There is no special requirements or procedures to execute `toppra` trajectories. 

If you have been able to execute trajectories on your robots, then do the same thing for the trajectories computed with toppra.

closed 2019-01-22 by hungpham2511