hungpham2511/toppra #25: Durations under 0.2 seconds not possible?

JadBatmobile, 2019-01-21:
I am using constant duration trajectory and desire trajectories with small durations,

I get this warning whenever i request a trajectory under 0.2 seconds

Desired duration 0.100000 seconds is not achievable. Returning the fastest parameterization with duration 0.200000 seconds

I am however providing acceleration and velocity limits that are extremely high, any ideas?

hungpham2511, 2019-01-22:
Can you try computing the time-optimal parametrization can see what is the optimal duration? We can start from there. 

Also, please try the most recent version of TOPPRA. In the past months, there are some updates related to numerical conditioning. 

If the bug persists, please assist by providing a script that contains both the part computing the parametrizations with the desired duration and the time-optimal one. I will happily take a look.

JadBatmobile, 2019-01-22:
Hello Hung,

After cloning the latest, it seems to have solved the problem!

It was definitely a problem with numerical conditioning. 

Thank you again, for a great package :+1: :100:

closed 2019-01-23 by hungpham2511