hungpham2511/toppra #27: phase-plane plot

GitLele92, 2019-01-25:
Is still possible using TOPP-RA to directly plot phase-plane? It was possible in TOPP but i cannot find the way to do this in TOPP-RA.
Thank you.

hungpham2511, 2019-01-26:
@GitLele92 Please take a look at this example. You obtain velocity-profile with `return_data` flag, then plot normally with `matplotlib`.

https://github.com/hungpham2511/toppra/blob/master/examples/kinematics.py#L39

GitLele92, 2019-01-28:
thank you. Sorry for the silly question,but once I have those data how do I have to organize the plot to get the same result of plotProfile of TOPP?

hungpham2511, 2019-01-29:
TOPP-RA is a different algorithm. Comparing with TOPP, it does not compute many things that plotProfile shows, such as the switch points, or the Velocity Limit Curves. 

So basically you can not get the same plot that plotProfile produces.

If you are interested in the difference, please have a look at the paper https://arxiv.org/abs/1707.07239