hungpham2511/toppra #29: The velocities start and end at non-zero values

tjh49, 2019-02-18:
Just run the basic example and find out the velocities are not zeros at beginning and ending. Wondering how come this happened since the initial and finals velocities should be zeros for robot controlling. 
[image]

hungpham2511, 2019-02-18:
Zero velocity can be achieved by setting `bc_type` to `clamped` in compute_trajectory.

See below for example.
jnt_traj, aux_traj, data = instance.compute_trajectory(0, 0, return_data=True, bc_type="clamped")

closed 2019-02-18 by hungpham2511

hungpham2511, 2019-02-18:
@tjh49 Hi, the lastest commit on branch master now automatically choose this setting if both start and end path velocities are zero.

tjh49, 2019-02-21:
problem solved! Thank you so much！

dtamino, 2024-11-08:
@hungpham2511 Is this still true today? There doesn't seem to be a `bc_type` argument to `compute_trajectory` anymore, and I'm seeing nonzero initial velocities, despite calling `compute_trajectory(0, 0)`. Looking at the code, I also don't see any logic to set `bc_type == 'clamped'` if the start and end velocities are zero.

dtamino, 2024-11-10:
I think I figured out what was wrong, and it was on my end: the velocities at start and end were in fact zero, but were increasing very quickly such that when sampling the output with not enough points, it looked like it wasn't zero. I found the code where the `bc_type` is set, it doesn't use `clamped` but does something equivalent (or better).

Instead, the issue I was seeing had to do with the interpolator used in input, `SplineInterpolator`, which being based on `CubicSpline` it can cause overshoots in the original path. I tried replacing it with one based on scipy's `PchipInterpolator` which guarantees monotonicity between points but that gave numerical errors in the solver ... anyways, not relevant to this issue, and well, I'll figure something out ;)