hungpham2511/toppra #31: TOPPRA for path planning

JadBatmobile, 2019-03-01:
Im curious about a potential use case of toppra. Say i have a path, parameterized by x, y, and theta. So a vehicle or robot is travelling some arbitrary path in the world, with each point being defined by an x,y coordinate, and a heading of the vehicle theta.

Would toppra work for time parameterizing this path, given that the units of the x,y coordinates are meters, and theta is radians? do the units have to be the same?

thank you

hungpham2511, 2019-03-01:
Unit should not be a problem, provided that the same unit is used for the constraints. As long as you only have bounds on velocity and acceleration, there should not be any problem.

If you do try to use toppra for that case, I am very interested in knowing how it works out.

Cheers,

JadBatmobile, 2019-03-01:
okay cool, i will let you know!!

JadBatmobile, 2019-03-01:
Hey hung,

when reading the paper, i am abit confused with equation (7).

I dont see how x(i+1) = x(i) + 2(s(i+1)-s(i))*(u(i))

hungpham2511, 2019-04-13:
I will close this issue. Feel free to open another one if need.

closed 2019-04-13 by hungpham2511