hungpham2511/toppra #32: Can TOPPRA solve the dynamic model considering friction?

Linjackffy, 2019-03-21:
If I want to consider Coulomb and Viscous friction into dynamic model:

M(q) * q_ddot + q_dot^T * C(q, q_dot) * q_dot + g(q) + Dv * q_dot + Fc *sign(q_dot) = torque

where M(q) , C(q)  and g(q)  are the same representation in tutorial of Second-Order Constraint.
Dv and Fc are coefficients of viscous and Coulomb friction, respectively.

Can TOPPRA solve it?

hungpham2511, 2019-03-21:
Unfortunately TOPPRA cannot solve problems with viscous friction. However, it can solve problems with dry friction.

Linjackffy, 2019-03-22:
I just try to add viscous and Coulomb friction in the Second-Order Constraint, TOPPRA return a result. From parametrization plot, it seems the given path is time-parameterizable, but the torque does not satisfy constraint. 
Anyway, thank for your reply.

hungpham2511, 2019-03-22:
Unfortunately, the **current implementation** of TOPPRA cannot handle dry friction. What I meant is TOPPRA as an algorithm can solve it, not this package.

 Should you wish to implement this functionality, I would be very happy to accept a pull request. Thanks!

Linjackffy, 2019-03-22:
TOPPRA is most stable and efficient algorithm for robot motion planning that I've ever used. It's my honor to contribute to this project. I would try to implement dry friction into TOPPRA.

closed 2019-04-13 by hungpham2511