hungpham2511/toppra #41: Final Trajectory Violates Acceleration Limits

EdsterG, 2019-04-15:
I've noticed that the optimized trajectories coming out of `instance.compute_trajectory` don't respect the acceleration limits. When I evaluate the trajectory at every point on `ss_waypoints`, at times the acceleration is almost double the limit.

EdsterG, 2019-04-15:
Was actually a bug in my code.

closed 2019-04-15 by EdsterG

cbames, 2019-05-15:
I'm running into a similar problem, what was the bug in your code?

mrunaljsarvaiya, 2019-07-11:
Any updates on this? I'm having the same problem

hungpham2511, 2019-07-12:
Can you post a minimal example that demonstrate the issue?

Bests,

Hung

mrunaljsarvaiya, 2019-07-12:
Thanks for the quick reply Hung! Here is a minimal example to demonstrate the issue.

Code and data for the test can be found here: [test_code](https://github.com/mrunaljsarvaiya/toppra/tree/sample_test/sample_test)

## Test description
The `toppra_test.npy` file contains data for a trajectory that has 384 points for a 7 DOF robot.
Most of the code in `toppra_example.py` is taken from the toppra tutorial [here](https://toppra.readthedocs.io/en/latest/tutorials/tut1_basic_example.html)

After sampling the generated trajectory to get the velocity and acceleration data, the maximum acc and vel per robot joint is found. Since the acc and vel limits are symmetric, the maximum acc and vel values are found by taking the max over their absolute values.

### Expected results
It is expected that the maximum velocity and acceleration per joint is lower than their corresponding limits

### Results
The results below show that the acceleration limits are not respected:

Joint 1: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.72, Max Acc: 1.02

Joint 2: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.57, Max Acc: 1.04

Joint 3: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.59, Max Acc: 1.30

Joint 4: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.62, Max Acc: 1.48

Joint 5: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.88, Max Acc: 1.40

Joint 6: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.85, Max Acc: 1.20

Joint 7: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.82, Max Acc: 1.41