hungpham2511/toppra #43: Implementation of the algorithm in C++?

oPiZiL, 2019-04-23:


hungpham2511, 2019-04-25:
I don't have any plan to do that in the near future. That said, the main difficulty is that there are no fast LP solvers for small-sized problems in C++, or that I couldn't find any.

JadBatmobile, 2019-08-15:
Any updates on this topic?

pbeeson, 2019-08-15:
I know at least one person that HAS done this (and even constrained through
jerk, not just acceleration), but it's not open source.  Perhaps they might
at least be willing to share which c++ solver they used.  I will ping them
to see.

JadBatmobile, 2019-08-16:
Thank you sir

pbeeson, 2019-08-16:
It looks like the programmer just rolled his own LP Solver for c++.  If I
hear more about this being licensed or opened, I'll follow-up on this issue.

JadTawil-theonly, 2019-11-26:
Hey Patrick,

Any updates on this ?

pbeeson, 2019-11-26:
Looks like the company he did this for is keeping this as private IP for
now.

JadTawil-theonly, 2019-11-26:
Thank you sir

oPiZiL, 2019-11-27:
> I don't have any plan to do that in the near future. That said, the main difficulty is that there are no fast LP solvers for small-sized problems in C++, or that I couldn't find any.

How about the glpk

EmanueleSiego, 2020-01-16:
> > I don't have any plan to do that in the near future. That said, the main difficulty is that there are no fast LP solvers for small-sized problems in C++, or that I couldn't find any.
> 
> How about the glpk

OSQP is a great open-source QP solver that my colleague is testing for his master thesis. I think it can be used for LP too.

hungpham2511, 2020-01-25:
Yes, both (glpk and osqp) are good solvers. However, since they are designed for problems having much larger sizes than those solved by `toppra`, there will be speed penalties.

zhaodong1986, 2020-03-06:
maybe could try Mike Hohmeyer's implementation for R. Seidel's LP solver(https://github.com/cgrushko/seidel-lp-solver) and tranlate it with C++

jmirabel, 2020-04-13:
> I don't have any plan to do that in the near future. That said, the main difficulty is that there are no fast LP solvers for small-sized problems in C++, or that I couldn't find any.

Hello,
In the article, you used qpOASES, which a C++ library. Any reason not to use it in the C++ implementation ?
I am thinking in implementing your algorithm.

hungpham2511, 2020-04-13:
Sure qpOASES is a really good solver, but it's not specialized to small LP problems. Thus it's unlikely that we would gain any speed benefits from implementing TOPPRA in C++ using qpOASES. I already use qpOASES Python wrapper in this version.

That said, having an interface in C++ is by itself a really nice thing already. Actually I have thought of doing that for a while but unfortunately didn't find the chance to do that.

jmirabel, 2020-04-13:
My primary goal is the C++ interface. The execution time is not a hard constraint for me.

jmirabel, 2020-04-13:
We can collaborate on this topic if you have some time for this.

hungpham2511, 2020-04-13:
In that case I don't see any problem with reimplementing toppra using qpoases as the underlying solver. 

> We can collaborate on this topic if you have some time for this.

That sounds great! Let's do that then.

hungpham2511, 2020-04-13:
@jmirabel Let's discuss the C++ interface design in this issue: https://github.com/hungpham2511/toppra/issues/82

hungpham2511, 2020-05-07:
An initial version is released in #100. Thus I will close this issue.

closed 2020-05-07 by hungpham2511