hungpham2511/toppra #44: the case that path-velocity enter zero

Linjackffy, 2019-04-26:
The figure below is the case that path-velocity enter zero. And no matter matter how large acceleration limits I give,  path-velocity always enter zero in the same position. 

[image]

The version of toppra I used is master branch. and code is below.

'''python

    import toppra as ta
    import toppra.constraint as constraint
    import toppra.algorithm as algo
    import numpy as np
    import matplotlib.pyplot as plt
    import time

    ta.setup_logging("INFO")

    def main():
    # Parameters
    N_samples = 5
    SEED = 9
    dof = 6

    # Random waypoints used to obtain a random geometric path. Here,
    # we use spline interpolation.
    # np.random.seed(SEED)
    in_file = 'input_point.txt'
    in_ = np.loadtxt(in_file)
    way_pts = np.array(in_).T
    path = ta.SplineInterpolator(np.linspace(0, 1, way_pts.shape[0]), way_pts, bc_type='not-a-knot')

    # Create velocity bounds, then velocity constraint object
    vlim_ = np.ones(dof) * np.pi * 4;
    vlim = np.vstack((-vlim_, vlim_)).T
    # Create acceleration bounds, then acceleration constraint object
    alim_ = np.ones(dof) * np.pi * 100;
    alim = np.vstack((-alim_, alim_)).T
    pc_vel = constraint.JointVelocityConstraint(vlim)
    pc_acc = constraint.JointAccelerationConstraint(
        alim, discretization_scheme=constraint.DiscretizationType.Interpolation)

    # Setup a parametrization instance. The keyword arguments are
    # optional.
    instance = algo.TOPPRA([pc_vel, pc_acc], path, np.linspace(0, 1, way_pts.shape[0]), solver_wrapper='seidel')

    # Retime the trajectory, only this step is necessary.
    t0 = time.time()
    jnt_traj, aux_traj, data = instance.compute_trajectory(0, 0, return_data=True, bc_type='clamped')
    # return_data flag outputs internal data obtained while computing
    # the paramterization. This include the time stamps corresponding
    # to the original waypoints. See below (line 53) to see how to
    # extract the time stamps.
    print("Parameterization time: {:} secs".format(time.time() - t0))

    ts_sample = np.linspace(0, jnt_traj.get_duration(), 100)
    qs_sample = jnt_traj.eval(ts_sample)  # sampled joint positions
    qds_sample = jnt_traj.evald(ts_sample)  # sampled joint velocities
    qdds_sample = jnt_traj.evaldd(ts_sample)  # sampled joint accelerations

    for i in range(dof):
        # plot the i-th joint trajectory
        plt.plot(ts_sample, qs_sample[:, i], c="C{:d}".format(i))
        # plot the i-th joint waypoints
        plt.plot(data['t_waypts'], way_pts[:, i], 'x', c="C{:d}".format(i))
    plt.xlabel("Time (s)")
    plt.ylabel("Joint position (rad/s^2)")
    plt.show()

    # Compute the feasible sets and the controllable sets for viewing.
    # Note that these steps are not necessary.
    _, sd_vec, _ = instance.compute_parameterization(0, 0)
    X = instance.compute_feasible_sets()
    K = instance.compute_controllable_sets(0, 0)

    X = np.sqrt(X)
    K = np.sqrt(K)

    plt.plot(X[:, 0], c='green', label="Feasible sets")
    plt.plot(X[:, 1], c='green')
    plt.plot(K[:, 0], '--', c='red', label="Controllable sets")
    plt.plot(K[:, 1], '--', c='red')
    plt.plot(sd_vec, label="Velocity profile")
    plt.title("Path-position path-velocity plot")
    plt.xlabel("Path position")
    plt.ylabel("Path velocity square")
    plt.legend()
    plt.tight_layout()
    plt.show()

    if __name__ == '__main__':
        main()

'''

way point is read form this file:
[input_point.txt](https://github.com/hungpham2511/toppra/files/3120671/input_point.txt)

hungpham2511, 2019-04-27:
The problem is that you select too few gridpoints. Normally the number of grid points should always be greater than the number of waypoints. Please try this:

     instance = algo.TOPPRA(
         [pc_vel, pc_acc], path, np.linspace(0, 1, 2000), solver_wrapper='seidel',
     )

You will find that path velocity is no longer zero.

Linjackffy, 2019-04-27:
Increasing gridpoints works for this case.
Another question is that why the curve of feasible_sets is not the same as Maximum Velocity Curve calculated by TOPP-IN?(the paths, constraints and gridpoints are all the same)
 
The figure below is Maximum Velocity Curve calculated by TOPP-IN. The green line is calculated by joint acceleration, the red line is calculated  by joint velocity.
[image]

hungpham2511, 2019-04-28:
The MVC should equals the square root of the feasible set's upper bounds.

Linjackffy, 2019-04-28:
I found the reason why sharp sink appears in the feasible set, the path is discontinuity in its first and second derivative. TOPPRA is correct.Thanks for your reply.

closed 2019-04-30 by Linjackffy