hungpham2511/toppra #51: gridpoints relevance in algorithm

JadBatmobile, 2019-04-29:
Hello, I am getting very different trajectories depending on the grippoint paramaeter in algo.TOPPRA.

Here is the code:

`def toppra_profiles(way_pts, am, vm):
	scalar = 1
	error_flag = True
	while scalar < 10000000000:
		way_pts2 = way_pts * scalar
		max_vel2 = vm * scalar
		max_acc2 = am * scalar

		path = ta.SplineInterpolator(np.linspace(0, 1, way_pts2.shape[0]), way_pts)
		vlim_ = np.ones(3) * max_vel2
		vlim = np.vstack((-vlim_, vlim_)).T
		# Create acceleration bounds, then acceleration constraint object
		alim_ = np.ones(3) * max_acc2
		alim = np.vstack((-alim_, alim_)).T
		pc_vel = constraint.JointVelocityConstraint(vlim)
		pc_acc = constraint.JointAccelerationConstraint(
			alim, discretization_scheme=constraint.DiscretizationType.Interpolation)
		instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=np.linspace(0, 1, 41),
							   solver_wrapper='hotqpoases')

		jnt_traj, aux_traj = instance.compute_trajectory(0, 0)

		if jnt_traj is not None:
			error_flag = False
			break
		else:
			print "trajectory is none profiles"

		scalar = scalar * 10

	if not error_flag:
		duration = jnt_traj.get_duration()
		ts_sample = np.linspace(0, duration, 100)
		jt = jnt_traj.eval(ts_sample)
		qs_sample = jnt_traj.evaldd(ts_sample)
		vel_sample = jnt_traj.evald(ts_sample)

		return jt, vel_sample, qs_sample, ts_sample, duration
	else:
		print "ERROR TRAJECTORY"
		return`

in the line `instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=np.linspace(0, 1, 41),
							   solver_wrapper='hotqpoases')`

when i use np.linspace(0, 1, 21), it gives much different results than np.linspace(0, 1, 51) for example.

here is the output for 3 joints for 51:

[image]

here is the output for 3 joints for 21:

[image]

The differences are drastic.
I would like to discuss the relevance of the grippoint parameter. After reading the paper, i am aware it is the discretization of the optimization space. Do you expect it to produce such differences?

hungpham2511, 2019-04-29:
What happens if you increase the number of grid points to 100? If you have too few grid points, the discretized problem will not reflect the actual problem correctly.

Bests,

Hung

JadBatmobile, 2019-04-30:
Yes when I increase the grid points past a certain number, it works well (trajectory topology is reasonable). I only wanted to discuss the theoretical aspects !

hungpham2511, 2019-04-30:
I see. What I think happens is that toppra will surely return a correct
answer for the discretized problem. However, if the grid points are too
sparse (20 pts), the discretized problem can no longer reflect the
continuous problem.

An interesting question is: What is the minimum number of points we can use
given a fixed error threshold (like velocity constraint satisfaction,
etc)... This is useful in providing an estimate of the computational
requirement for a given problem instance.

closed 2019-12-08 by hungpham2511