hungpham2511/toppra #53: python requirements boken in latest 16.04 updates

pbeeson, 2019-05-18:
As of 5/17/2019, Ubuntu apt updates to python-apt packages have broken the toppra requirements install.  numpy is no longer supported with python2, so moving from requirements.txt to requirements3.txt got me through the requirements install but then building toppra yells about cython.

```Errors     << toppra:make /home/pbeeson/projects/pharaoh/ros/logs/toppra/build.make.034.log          
You are using pip version 8.1.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Traceback (most recent call last):
  File "setup.py", line 2, in <module>
    from Cython.Distutils import build_ext
ImportError: No module named Cython.Distutils
Traceback (most recent call last):
  File "setup.py", line 2, in <module>
    from Cython.Distutils import build_ext
ImportError: No module named Cython.Distutils
make[2]: *** [CMakeFiles/setup] Error 1
make[1]: *** [CMakeFiles/setup.dir/all] Error 2
make: *** [all] Error 2
```

hungpham2511, 2019-05-18:
I have tested toppra on Python 3 and things seem yo work. Can I know the steps you have taken to install?

Btw you can try 'pip install toppra' instead.

Bests,

Hung

pbeeson, 2019-05-18:
It does work for python3.  I got that working eventually.

But I can’t pip install toppra because your latest changes break some
benchmarks tests.  Changes you made prior to the auto scaling feature are
actually more stable than later versions in my testing.

closed 2019-05-18 by pbeeson