hungpham2511/toppra #55: Spike in the Reachable set

tombelv, 2019-07-13:
I'm trying to use toppra to compute the optimal parametrization and joint torques for a cartesian elliptical task. I already have the joint trajectory computed inverting the kinematics and I'm feeding this to toppra, with joint velocity and joint torque bounds.

The path is given by 1000 equally spaced waypoints, with 5000 gridpoints (I've also tried with much more waypoints, less gridpoints and so on with no success)

While no error is shown, something seems to be wrong, in particular seeing the spike at the beginning of the feasible set. The joint torque and velocity are satisfying their limits, hinting that there should not be an error in the setting of the constraints.

I've tried both the qpoases and seidel solvers with no difference, as well as the scaling option.

Moreover, I'm using the same framework from a paper from my university which has solved the same problem in the past using TOPP, which managed to get a minimum time of 1 second, while I get 2 seconds.
[image]

Below is the same phase plane image from the article I'm following, that used TOPP
[image]

Do you have any suggestion on what could be causing this problem and how to proceed?

This is my code if it can help
`path = ta.SplineInterpolator(time, joint_pos[:,:num_dof-1])

vel_lim = np.array([1.92, 1.92, 2.23, 2.23, 3.56, 3.21])
vel_lim_stack = np.vstack((-vel_lim, vel_lim)).T

pc_vel = constraint.JointVelocityConstraint(vel_lim_stack)

def inverse_dynamics(q, dq, ddq):
    q = np.append(q, 0)
    dq = np.append(dq, 0)
    ddq = np.append(ddq, 0)
    tau = inv_dyn(q, dq, ddq)

    return tau[:num_dof-1]

def F(q):
    return np.vstack((np.eye(num_dof-1), -np.eye(num_dof-1)))

def g(q):
    return np.append(torque_lim, torque_lim)

pc_torque = constraint.SecondOrderConstraint(inverse_dynamics, F, g, dof=num_dof-1)

gridpoints = np.linspace(0, path.duration, 5000)

instance = algo.TOPPRA([pc_vel, pc_torque], path, gridpoints=gridpoints, solver_wrapper='qpoases')
jnt_traj, aux_traj, int_data = instance.compute_trajectory(0, 0, return_data=True)

X = instance.compute_feasible_sets()`

hungpham2511, 2019-07-13:
The spike at the beginning of the controllable set is not an issue. Often its because of the interpolation scheme.

About the comparison with TOPP, most likely yhe

Bests,

Hung

[Figure_1]<https://user-images.githubusercontent.com/43341058/61173672-b8771480-a596-11e9-8c9d-de3d8f4f113e.png>

[image]<https://user-images.githubusercontent.com/43341058/61173792-75b63c00-a598-11e9-9272-090306779d39.png>

tombelv, 2019-07-13:
Thank you for the quick response, but maybe you have sent without finishing the sentence about TOPP?

About the interpolation scheme, are you suggesting to change it?

hungpham2511, 2019-07-14:
Well, I do not suggest changing the interpolation scheme unless you have a reason to do so. In any case toppra does not assume any specific interpolation scheme, so it really depends on your settings and usecase.

About the comparision between TOPP and TOPPRA, look like there is a discrepancy in the way the problem is formulated. I recommend plotting the joint trajectory to debug.

tombelv, 2019-07-14:
I understand, thank you.
If it is ok, I would leave the issue open while I try to understand more what is going on, maybe some discussion could arise. I will close it if I can't manage to advance or I find the cause.

tombelv, 2019-08-11:
Closing since while the spike still appears it varies when I change some parameters, I couldn't isolate the cause but there seems to be not much too add.
Thank you for the support.

closed 2019-08-11 by tombelv