hungpham2511/toppra #56: Trajectory Violates Constraints

mrunaljsarvaiya, 2019-07-15:
I've been trying to use toppra to generate a trajectory for a 7 DOF robot but the trajectories generated violate the velocity and acceleration limits. 
Note that this is issue follows the conversation from #41 

Here is a minimal example to demonstrate the issue.
Code and data for the test can be found [here](https://github.com/mrunaljsarvaiya/toppra/tree/sample_test/sample_test)

## Test description
The `toppra_test.npy` file contains data for a trajectory that has 384 points for a 7 DOF robot.
Most of the code in `toppra_example.py` is taken from the toppra tutorial [here](https://toppra.readthedocs.io/en/latest/tutorials/tut1_basic_example.html)

After sampling the generated trajectory to get the velocity and acceleration data, the maximum acc and vel per robot joint is found. Since the acc and vel limits are symmetric, the maximum acc and vel values are found by taking the max over their absolute values.

### Expected results
It is expected that the maximum velocity and acceleration per joint is lower than their corresponding limits

### Results
The results below show that the limits are not respected:

Joint 1: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.72, Max Acc: 1.02

Joint 2: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.57, Max Acc: 1.04

Joint 3: 
Vel limit: 0.87, Acc limit: 0.8
Max Vel: 0.59, Max Acc: 1.30

Joint 4: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.62, Max Acc: 1.48

Joint 5: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.88, Max Acc: 1.40

Joint 6: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.85, Max Acc: 1.20

Joint 7: 
Vel limit: 0.87, Acc limit: 1.0
Max Vel: 0.82, Max Acc: 1.41

hungpham2511, 2019-07-15:
Hi @mrunaljsarvaiya, thanks for using toppra.

The reason why toppra doesn't work as expected is that the input path is way too noisy. toppra requires a certain level of smoothness fo function properly.     
    
[image]

This is the input file you throw to the algorithm. I would suggest you to try smoothing the path and try again. Thanks!

mrunaljsarvaiya, 2019-07-15:
Is there a way to tell that the solver failed to produce a trajectory that meets the constraints without having to check the data ?

hungpham2511, 2019-07-16:
Very good question. I am sure there is a way, but I have not really thought about this in detail. Let me re-open this issue just to keep track of ideas for this new feature.

Any thoughts?

mrunaljsarvaiya, 2019-07-16:
I'm not too familiar with the theory behind toppra, why does the algorithm require a smooth trajectory to work? I would have expected that if it sees a large spike/dip, a slower trajectory would be generated in that segment.

hungpham2511, 2019-07-16:
Perhaps you want to take a look at the paper at
https://arxiv.org/abs/1707.07239

mrunaljsarvaiya, 2020-03-12:
I have also noticed that if trajectory generated by toppra is rerun through toppra, it produces a faster trajectory that deviates less from the limits. If this "feedback" loop is repeated 3 times, the trajectory gets faster and better abides by the limits with every iteration

closed 2020-06-08 by github-actions[bot]