hungpham2511/toppra #59: Optimization not always working

pbeeson, 2019-07-30:
I've seen examples where toppra (both my "old" fork and the newest upstream) provide the same wrong answer in terms of time for the last waypoint in a trajectory.  HEre is an example:
6 waypoints (6-DOF robot):
```
[[-0.030642413736830083, 0.9210788610725734, 1.1005370198753621, 0.0010817886435241263, 1.1229495518273334, 2.8986148215009915], [-0.030763529996659454, 0.9226167221835322, 1.098539142483565, 0.001086064487964933, 1.1234197081586665, 2.9100717970800467], [-0.03088464625648882, 0.9241545832944911, 1.0965412650917683, 0.00109034033240574, 1.12388986449, 2.921528772659102], [-0.03100576251631819, 0.9256924444054498, 1.0945433876999713, 0.0010946161768465468, 1.1243600208213334, 2.932985748238157], [-0.030993973983559547, 0.9343914313305473, 1.0990185656991058, 0.0010486544290009307, 1.1112167154104744, 2.9329954455064784], [-0.03098218545080091, 0.9430904182556445, 1.1034937436982406, 0.0010026926811553146, 1.0980734099996154, 2.9330051427747996]],
           [0.785398, 0.6283190000000001, 0.8901180000000001, 0.942478, 0.942478, 1.50796],
           [78.5398, 62.831900000000005, 89.0118, 94.2478, 94.2478, 150.796]
```

The last two lists above are the max velocities/accelerations.

If I run this through toppra, I get a final times of:  `[0.0, 0.018758242674088654, 0.027433411317876365, 0.037597795704716795, 0.06504036006938992, 53.71761885739489]`.

That last time is "crazy".  If I simply do waypoints.append(waypoints[-1]) in order to make a 7 point trajectory where the last value is repeated, I get something more reasonable: `
[0.0, 0.018760785795569175, 0.027481922635328733, 0.037832897511885226, 0.06618539826249735, 0.09915751384422705, 0.13215386747421617]`

hungpham2511, 2019-07-31:
Most likely this is because of the interpolation function. My guess is that the interpolated path (from these input waypoints) is oscillating wildy.

pbeeson, 2019-07-31:
I'm unsure if you are trying to say that this is an issue with CubicSplie or with how I'm setting up the problem.  My interpolated waypoints are simply defined as: 
```
ws = np.linspace(0, 1, len(way_pts))
path = toppra.SplineInterpolator(ws, [list(x) for x in way_pts])
.....
instance = toppra.algorithm.TOPPRA([pc_vel, pc_acc], path, solver_wrapper='seidel', gridpoints=gp)
```

hungpham2511, 2019-07-31:
Yes I meant it is the problem with the Cubic Spline interpolation.

Bests,

Hung

hungpham2511, 2019-08-03:
Actually I run your example and got a fairly reasonable result. 
[image]

pbeeson, 2019-08-03:
That’s certainly not what I am seeing.  If you can post the example program
you write, I can look to see what differences in setup there might be.
Likewise I will post an example program showing my results as soon as I can.

hungpham2511, 2019-08-03:
I just modify the kinematics.py example. Please find the code below:
```
import toppra as ta
import toppra.constraint as constraint
import toppra.algorithm as algo
import numpy as np
import matplotlib.pyplot as plt
import time

ta.setup_logging("INFO")

def main():
    # Parameters
    N_samples = 5
    SEED = 9
    dof = 7

    # Random waypoints used to obtain a random geometric path. Here,
    # we use spline interpolation.
    np.random.seed(SEED)
    way_pts = np.array([[-0.030642413736830083, 0.9210788610725734, 1.1005370198753621, 0.0010817886435241263, 1.1229495518273334, 2.8986148215009915], [-0.030763529996659454, 0.9226167221835322, 1.098539142483565, 0.001086064487964933, 1.1234197081586665, 2.9100717970800467], [-0.03088464625648882, 0.9241545832944911, 1.0965412650917683, 0.00109034033240574, 1.12388986449, 2.921528772659102], [-0.03100576251631819, 0.9256924444054498, 1.0945433876999713, 0.0010946161768465468, 1.1243600208213334, 2.932985748238157], [-0.030993973983559547, 0.9343914313305473, 1.0990185656991058, 0.0010486544290009307, 1.1112167154104744, 2.9329954455064784], [-0.03098218545080091, 0.9430904182556445, 1.1034937436982406, 0.0010026926811553146, 1.0980734099996154, 2.9330051427747996]])
    path = ta.SplineInterpolator(np.linspace(0, 1, len(way_pts)), way_pts)

    vlim_ = np.r_[0.785398, 0.6283190000000001, 0.8901180000000001, 0.942478, 0.942478, 1.50796]
    alim_ = np.r_[78.5398, 62.831900000000005, 89.0118, 94.2478, 94.2478, 150.796]

    # Create velocity bounds, then velocity constraint object
    vlim = np.vstack((-vlim_, vlim_)).T
    # Create acceleration bounds, then acceleration constraint object
    alim = np.vstack((-alim_, alim_)).T
    pc_vel = constraint.JointVelocityConstraint(vlim)
    pc_acc = constraint.JointAccelerationConstraint(
        alim, discretization_scheme=constraint.DiscretizationType.Interpolation)

    # Setup a parametrization instance
    instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=np.linspace(0, 1, 101),
                             solver_wrapper='seidel')
    t0 = time.time()
    # Retime the trajectory, only this step is necessary.
    jnt_traj, aux_traj = instance.compute_trajectory(0, 0)
    print("Parameterization took {:} secs".format(time.time() - t0))
    ts_sample = np.linspace(0, jnt_traj.get_duration(), 100)
    qs_sample = jnt_traj.evald(ts_sample)

    plt.plot(ts_sample, qs_sample)
    plt.xlabel("Time (s)")
    plt.ylabel("Joint acceleration (rad/s^2)")
    plt.show()

    # Compute the feasible sets and the controllable sets for viewing.
    # Note that these steps are not necessary.
    X = instance.compute_feasible_sets()
    K = instance.compute_controllable_sets(0, 0)

    _, sd_vec, _ = instance.compute_parameterization(0, 0)

    X = np.sqrt(X)
    K = np.sqrt(K)

    plt.plot(X[:, 0], c='green', label="Feasible sets")
    plt.plot(X[:, 1], c='green')
    plt.plot(K[:, 0], '--', c='red', label="Controllable sets")
    plt.plot(K[:, 1], '--', c='red')
    plt.plot(sd_vec, label="Velocity profile")
    plt.title("Path-position path-velocity plot")
    plt.xlabel("Path position")
    plt.ylabel("Path velocity square")
    plt.legend()
    plt.tight_layout()
    plt.show()

if __name__ == '__main__':
    main()
```

pbeeson, 2019-08-03:
OK, so the difference is that for gridpoints, instead of 101, we use
len(waypts).   Because sometimes we are dealing with trajectories on the
order of 5000 points (And sometimes as few as 3) , a constant of 101
doesn't seem appropriate.  I don't recall if I needed len(waypts) for
trajectories > 101 waypoints long or not.   Any insight into how gridpoints
might be assigned programmatically to work for any number of waypoints
would be useful.

hungpham2511, 2019-08-04:
Intuitive the number of grid points should be sufficient to capture intrinsic properties of the trajectory. Even if you have only 3 waypoints, but if those vary significantly from each other, a good number of grid points should still be selected. 10 gridpoints is almost always too small.

Mathematically I am sure you can look at the derivatives of the input path to make a guess at what should the grid-points be.

pbeeson, 2019-08-04:
I don’t fully follow your explanation.  If you could re-phrase or be more
specific, maybe I’ll get it the second time.

hungpham2511, 2019-08-05:
Please see this function: https://github.com/hungpham2511/toppra/blob/bc0fe18ce87d41195546515c8e483cf27616c015/toppra/interpolator.py#L49

pbeeson, 2019-08-13:
This seems to work for now:
```
def propose_gridpoints(gp, path, delta=1e-4):
    """"""
    gridpoints_ept = gp
    while True:
        updated = False
        for idx in range(len(gridpoints_ept) - 1):
            gp_mid = 0.5 * (gridpoints_ept[idx] + gridpoints_ept[idx + 1])
            dist = gridpoints_ept[idx + 1] - gridpoints_ept[idx]
            max_err = np.max(np.abs(0.5 * path.evaldd(gp_mid) * dist ** 2))
            if max_err > delta:
                gridpoints_ept.append(gp_mid)
                updated = True
        if updated:
            gridpoints_ept = sorted(gridpoints_ept)
        else:
            break

    return gridpoints_ept
```

Called by 
```
ws = np.linspace(0, 1, len(way_pts))
gp = propose_gridpoints(list(ws), path)
```

closed 2019-08-16 by hungpham2511

pbeeson, 2019-08-16:
I would propose that perhaps you add the propose_gridpoint() function into
toppra whenever gridpoints sent is are None.  Currently I believe it
defaults to 100 gridpoints.

hungpham2511, 2019-08-16:
Thanks for the recommendation. That's indeed what I intend to do.