hungpham2511/toppra #82: C++ Implementation

hungpham2511, 2020-04-13:
We will use this issue to flesh out some key designs of the C++ interface.

There are a few core components that are required.
- [ ] Trajectory representation
- [ ] LP solver wrapper;
- [ ] Core TOPPRA pipeline;

I am most inexperienced with trajectory representation in C++. Use OpenRAVE before with its representation, but bringing it over is a huge dependency that I am not willing to make.

jmirabel, 2020-04-13:
One extra mention: [Pinocchio](https://github.com/stack-of-tasks/pinocchio) could also be used to interpolate the trajectories. It does the interpolation in Lie groups.

hungpham2511, 2020-04-13:
Totally agree on the abstract base class. I propose we make use of the current Python interfaces. Have a look:
- constraints: https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/constraint/linear_constraint.py#L7
- trajectory/geometric path:
https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/interpolator.py#L84
- LP/QP solver wrapper interface
https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/solverwrapper/solverwrapper.py#L42

The main toppra algorithm is relatively straightforward once we have all the pieces working.

Regarding license, I propose to use the MIT license, the same as the Python codebase. What do you think?

Another point is testing, I would like the code to have good unit test coverage. Shall we use google-test?

jmirabel, 2020-04-13:
> * constraints: https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/constraint/linear_constraint.py#L7

Any reason not to use this one:
https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/constraint/constraint.py#L30

> Regarding license, I propose to use the MIT license, the same as the Python codebase. What do you think?

MIT is fine for me.

> Another point is testing, I would like the code to have good unit test coverage. Shall we use google-test?

I almost never used google-test but I guess it isn't hard to use.

hungpham2511, 2020-04-14:
> Any reason not to use this one:
> https://github.com/hungpham2511/toppra/blob/c013302cf78fe369c61a293789ee32cac0f4f759/toppra/constraint/constraint.py#L30
>
>
> MIT is fine for me.
>
>
> I almost never used [...]

`LinearConstraint` has a more concrete signature, and covers a good range of constraints. The more abstract `Constraint` is to provide support for other forms of constraints (such as conic, etc). These constraints can't be handled by qpOASES anyway.

Feel free to open PRs to `develop` branch directly whenever you have something. By today I will set up a skeleton for the cpp code. Please let me know what you will be working on in this thread so that we don't work on the same thing.

jmirabel, 2020-04-14:
> Feel free to open PRs to `develop` branch directly whenever you have something. By today I will set up a skeleton for the cpp code. Please let me know what you will be working on in this thread so that we don't work on the same thing.

Ok.

> Edit: I see that https://github.com/stack-of-tasks/pinocchio is header-only. Would you recommend that approach? Not entirely sure of the advantages/disadvantages myself.

Pinocchio is a lighter dependency than OpenRAVE. It is close to be minimalist for the need of TOPP-RA. The torque limit constraint can be easily implemented using [RNEA](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a33f3f128c9cf62407691acaedd28e68d).
A good option for this library regarding torque limits is:
- provide an abstract class with a pure virtual method `computeTorque(q, v, a)` and
- provide an implementation of the torque limits using Pinocchio that can be enabled / disabled with a CMake flag.

hungpham2511, 2020-04-14:
I have the boilerplate setup. A tentative implementation plan to get the library to a decent state could consist of:
- [ ] Abstract path interface;
- [ ] A simple polynomial path class (for testing);
- [ ] Abstract constraint interface;
- [ ] Simple acceleration/velocity constraints;
- [ ] Torque constraints (pinocchio, openrave);
- [ ] Wrapper for qpOASES;
- [ ] TOPPRA algorithm;

I will be working on the first two this week.

jmirabel, 2020-04-14:
Thank you. I will start working on the constraints by tomorrow.
*not this time* mean *left for future developments* ?

quangounet, 2020-04-17:
@jmirabel For TOPP in Lie groups we have a paper (https://www.ntu.edu.sg/home/cuong/docs/TOPPSO3SE3.pdf) and an implementation (https://github.com/dinhhuy2109/TOPP-SO3). You could perhaps re-use some derivations.

jmirabel, 2020-04-17:
@quangounet Thanks for the pointer. I think I read this article when I implemented splines in Lie groups in HPP.

jmirabel, 2020-04-17:
By the way, @quangounet, when I implemented the splines, I first looked for how to compute weighted centroids in SO(3) because that would be a more appropriate way to have splines in SO(3) (rather than in so(3), then integrated). I couldn't find anything (without optimization). Would you have any pointer on this ?

jmirabel, 2020-04-21:
How do we organize ourselves for implementation of TOPPRA algorithm ?

hungpham2511, 2020-04-22:
The TOPPRA algorithm is quite straightforward to implement given the available solver wrappers. I can implement something quite quickly. I would probably use the same interface as the Python implementation, what do you think?

https://github.com/hungpham2511/toppra/blob/24ab163c1999e04c02075919145351ae140f370e/toppra/algorithm/algorithm.py#L14

The Python impl has a provisional double inheritance for non-TOPPRA algorithm. Sadly it is never used. For this C++ impl, I think we only need a single abstract interface and two derived classes: TOPPRA and TOPPRAsd (specified duration). Other variants (smoothing, tracking etc) can then be added in by subclassing.

Another thing we need to do is recovery of the final trajectory once a parametrization is available, as in here.

https://github.com/hungpham2511/toppra/blob/24ab163c1999e04c02075919145351ae140f370e/toppra/algorithm/algorithm.py#L89

Also, more tests for the solver wrappers/constraints are always good.

Linjackffy, 2020-04-22:
@jmirabel manif maybe a good choice for implementing splines in Lie groups. It provides many useful curves like bspline and bezier.

jmirabel, 2020-04-22:
@Linjackffy What is *manif* ?

jmirabel, 2020-04-22:
About the architecture of the TOPPRA algorithm, yes we can keep the current structure. I would only put the mechanism to find one available solver either outside the class or in a static function but not in the constructor, and let the constructor simply take a (pointer to a ?) solver.

Linjackffy, 2020-04-22:
@jmirabel You could look at this webpage: [manif](https://github.com/artivis/manif)

jmirabel, 2020-04-22:
Thanks. I know personally the authors of this work. It is an interesting work. I think this package was created after I implemented splines in Lie groups.

jmirabel, 2020-04-29:
The main algorithm has been implemented so I think this issue can be closed. Bug reports or feature requests can go into their own specific issue.
@hungpham2511 if you agree, I let you close.

closed 2020-04-29 by hungpham2511