hungpham2511/toppra #95: Build cpp doc on readthedocs with sphinx

hungpham2511, 2020-04-25:


jmirabel, 2020-04-25:
This will take you a lot of effort for not so much. readthedocs does not provide Doxygen support and getting sphinx to generate your C++ documentation is not gonna be easy.

At the moment, getting the documentation is done by:
- cloning the repo (but if you are looking at the API, this is not a big problem) and
- running `make doc` (this means you have to run cmake).

Is this method a problem ?

hungpham2511, 2020-04-25:
This is to serve both the Python and C++ API in the same place. https://toppra.readthedocs.io/en/latest/

I have done something similar before using sphinx via breath (https://breathe.readthedocs.io/en/latest/directives.html). As long as the readthedocs server can run 'doxygen', there should be no problem.

hungpham2511, 2020-10-29:
@jmirabel I am thinking that the doxygen docs build can be uploaded to the same place as the sphinx docs at https://hungpham2511.github.io/toppra/ 

What do you think?

jmirabel, 2020-10-29:
It would be a nice indeed.