hungpham2511/toppra #101: Cartesian acceleration/velocity constraints

hungpham2511, 2020-05-07:
**Describe the solution you'd like**
Implementations of Cartesian acceleration/velocity constraints with the C++ API

thummel42, 2020-08-13:
Hey just asking if there is an example python interface that can be ported to cpp for this. Specifically would like to do varying velocity and acceleration constraints.

hungpham2511, 2020-08-13:
Yes I plan to do this for sometime for unfortunately couldn't find the time for it. It should be a straightforward extension of https://github.com/hungpham2511/toppra/blob/develop/toppra/constraint/linear_second_order.py

stevegolton, 2020-10-10:
Hi @hungpham2511. I had a go at making a Cartesian velocity constraint class (see listing below). It seems to work just fine using an `fk` implementation based on PyKDL's `ChainFkSolverVel_recursive`, but I'm not sure why I have to square the outputs of the `invdyn` and `constraintg` functions. I'll admit my understanding of the maths is hazy. Is this something like the class you had imagined?

```
class CartesianVelocityConstraint(constraint.SecondOrderConstraint):
    """This class implements a constraint on Cartesian linear and angular velocity of some link in the robot chain.
    """

    def __init__(self, fk, max_linear, max_angular, dof):
        """Initialize the constraint.

        Parameters
        ----------
        fk: (np.ndarray, np.ndarray) -> (linear_vel, angular_vel)
            The "FK" function that receives joint position and velocity as inputs and ouputs
            the linear and angular velocity of some link of the robot.
        max_linear: float
            The max linear velocity allowed for that link.
        max_angular: float
            The max angular velocity allowed for that link.
        dof: int
            The dimension of the joint position.
        """
        super(CartesianVelocityConstraint, self).__init__(self.invdyn, self.constraintf, self.constraintg, dof)
        self.fk = fk
        self.max_linear = max_linear
        self.max_angular = max_angular

    def invdyn(self, q, dq, ddq):
        linear, angular = self.fk(q, dq)
        return np.array([linear**2, angular**2])

    def constraintf(self, q):
        # 2x2 identity matrix
        return np.identity(2)

    def constraintg(self, q):
        return np.array([self.max_linear**2, self.max_angular**2])
```

hungpham2511, 2020-10-11:
Hi @stevegolton, have you had a look at this [example](https://github.com/hungpham2511/toppra/blob/develop/examples-old/cartesian_accel.py)? By the way, if you have a PR for this feature I will be happy to review and merge it.

stevegolton, 2020-10-12:
Thanks! I hadn't found that example before, that's really helpful. PR submitted #144.

jmirabel, 2020-10-28:
For completeness, I see two ways of making velocity limit fit into the constraint model of TOPPRA.
1. limit on the squared S-norm (See #146)
2. limit on individual components that ends up to ` sd^2 < (limit / v_i)^2` where `v_i` is the i-th component of `v = J(q(s)) qdot(s)`

jmirabel, 2020-10-29:
> I'm not sure why I have to square the outputs of the `invdyn` and `constraintg` functions.

The algo works by solving a bunch of LPs. The variables are `\ddot{s}` and `\dot{s}^2`. Your constraint has to be linear with respect to these variables.

PR #146 (2020-11-05): [cpp] Add CartesianVelocityNorm and an implemention with Pinocchio
Partially fixes #101 and is related to #144

From the doxygen doc:
[image]
closed 2020-11-05 by jmirabel