hungpham2511/toppra #102: Generate output trajectory

hungpham2511, 2020-05-07:
**Describe the solution you'd like**
A feature that allows to "plug" in the new time parametrization to the original geometric path and outputs:
- joint pos/vel/accel at given time instances;

This can perhaps be done by adding a another method to the GeometricPath class, e.g. 'applyParametrization', to obtain the new trajectory.

hungpham2511, 2020-05-14:
A prototype in Python 
https://github.com/hungpham2511/toppra/blob/ba9328cd2f4e0ecb1eb71bba5c98fab78a8b4be6/toppra/parametrize_const_accel.py#L8

hungpham2511, 2020-06-12:
The parametrizer can additionally implement some form of smoothing on the velocity profile. One way is to optimize the variables (sd, s) such that some measure of jerkiness is minimized.

thummel42, 2020-06-16:
Question about the python prototype (I am converting it to C++). It requires both the path velocities and squared path velocities, however the C++ implementation of the algorithm only supplies the squared path velocities as part of ParamaterizationData. Is there a missing data field or am I misinterpreting the output of the algorithm

hungpham2511, 2020-06-17:
As the names suggest, you only need to compute the square root to obtain what you want

thummel42, 2020-06-17:
Ok cool, I assumed that. I just wanted to make sure the loss of sign of
velocity is ok.

hungpham2511, 2020-06-17:
No worry, the path velocity is always non-negative so you can take square root without worrying about the sign.

ndehio, 2020-09-13:
Hi there,
I tried to use the

ndehio, 2020-09-13:
Hi there,
is there any update on this issue?
I tried to use the result of the TOPPRA algorthim as input for the c++ trajectory parametrizer implemented in the branch `feat-parametrizer`. For this, I implemented a cpp-test that follows the python-example described [here](https://hungpham2511.github.io/toppra/quickstart.html#time-optimal-retime-with-kinematics-constraints) and submitted a [pull-request](https://github.com/hungpham2511/toppra/pull/140).

There is still a minor issue, but I guess this can be solved easily with your expert advise.

PR #129 (2020-09-19): [cpp]Implement trajectory parametrizer
Fixes #102 

Changes in this PRs:
- [x] Constant Accel parametrizer
- [x] Unit tests

Checklists:
- [x] Update CHANGELOG.md with a single line describing this PR
closed 2020-09-19 by hungpham2511