hungpham2511/toppra #109: Implement Seidel LP solver for C++

hungpham2511, 2020-05-14:
The Cython implementation of Seidel LP solver 

https://github.com/hungpham2511/toppra/blob/ba9328cd2f4e0ecb1eb71bba5c98fab78a8b4be6/toppra/solverwrapper/cy_seidel_solverwrapper.pyx#L65

is still the fastest solver for toppra. This solver was very important to achieve sub-millisecond solution time, or, a reasonable time for large and difficult problem.

It will be ideal to implement this in C++, then provide a binding back to the Python API.

yy16808, 2020-07-15:
How to call  Cython implementation in C++?

hungpham2511, 2020-07-15:
Hi, I have no idea how to do that, and even if it's possible I doubt it's a good idea.

Bests,

Hung

xiaodaxia-2008, 2020-10-19:
Maybe a re-implementation with cpp is fast enough. Actually I've already done this part after trying several third-party solvers (which are more generous but not fast enough for 2D LP).

closed 2021-01-14 by hungpham2511