hungpham2511/toppra #112: [C++] numerical issue

jmirabel, 2020-05-28:
Hi @hungpham2511 

I encounter a numerical issue at this line:
https://github.com/hungpham2511/toppra/blob/5ffa8e1b92c8151e837a2fd4253948860d9bbc38/cpp/src/toppra/solver/glpk-wrapper.cpp#L113

Function `intersection(a, b, c)` computes the intersection between interval a and interval b and stores it into c. At some point, I get `a = [0.11138810763690521, 0.11138810763690521 ]` and `b = [ 0, 0.11138810763690514 ]`. If you look carefully at the values, you will see that:
- the intersection is empty and
- the upper bound of `b` is very close to `a`.

I get this during the forward pass. In my scenario, I have only one linear velocity constraint (which gives `b`). When the fault happens, the `a` value is the `x` value passed to the solver (so the feasible set at the index `i`).

I can make a work around in this solver but I am wondering whether this should be addressed at a higher stage, like in the TOPPRA algorithm itself.

Any thoughts on this ?

jmirabel, 2020-05-28:
FYI: I also have issues with qpOASES. I did not look at it as closely so I cannot say whether it is the same thing.

jmirabel, 2020-05-28:
The following patch fixes the issue in both solver:
```
diff --git a/cpp/src/toppra/solver/glpk-wrapper.cpp b/cpp/src/toppra/solver/glpk-wrapper.cpp
index b8c52eb..f27e481 100644
--- a/cpp/src/toppra/solver/glpk-wrapper.cpp
+++ b/cpp/src/toppra/solver/glpk-wrapper.cpp
@@ -19,7 +19,7 @@ int bnd_type(const value_type& l, const value_type& u)
     if (u == infty) return GLP_FR;
     else return GLP_UP;
   } else {
-    if (l == u) return GLP_FX;
+    if (l >= u) return GLP_FX;
     else if (u == infty) return GLP_LO;
     else return GLP_DB;
   }
diff --git a/cpp/src/toppra/solver/qpOASES-wrapper.cpp b/cpp/src/toppra/solver/qpOASES-wrapper.cpp
index e10135b..10efb99 100644
--- a/cpp/src/toppra/solver/qpOASES-wrapper.cpp
+++ b/cpp/src/toppra/solver/qpOASES-wrapper.cpp
@@ -118,6 +121,9 @@ bool qpOASESWrapper::solveStagewiseOptim(std::size_t i,
   //    H, g, self._A, l, h, self._lA, self._hA, np.array([1000])
   //)
   int nWSR = 1000;
+  // Make sure bounds are correct
+  h = h.cwiseMax(l);
+  m_hA = m_hA.cwiseMax(m_lA);
   if (H.size() == 0) {
     m_impl->qp.setHessianType(qpOASES::HST_ZERO);
     res = m_impl->qp.init (NULL, g.data(),

```

However, this patch prevent the solver to catch infeasibility...

hungpham2511, 2020-05-28:
I have faced issue with infeasibility in the forward pass in the Python implementation as well. This is definitely a numerical issue because mathematically the forward pass is always feasible.

In the Python implementation I resolve this by doing a line-search in the forward pass instead of solving the 2D program. See https://github.com/hungpham2511/toppra/blob/5ffa8e1b92c8151e837a2fd4253948860d9bbc38/toppra/solverwrapper/hot_qpoases_solverwrapper.py#L175  Actually this is even better because the line search is much simpler to solve.

jmirabel, 2020-05-28:
Thanks for your insight. It is helpful.

>  Actually this is even better because the line search is much simpler to solve.

I agree the problem is simpler to solver but I expect the solver to actually do that simplification:
- GPLK knows when a variable is fixed,
- I am not sure for qpOASES but as equality constraints are built from inequalities, I would expect it to detect fixed variables.

Did you check that it is actually faster ?

hungpham2511, 2020-05-28:
Yes for qpOASES. I measured it. Also, the expected time-complexity of 2D LP is O(N^2), while 1D LP is O(N). So the speedup is quite minimal for fewer constraints. Nothing very major. 

That said if we really want to get the best possible performance out of the algorithm, then each LP should also use information from the last, i.e., what constraints are active in the last step. Since we know for sure that there can be a fixed number of constraint-switching in the final trajectory, the time-complexity is actually ~ O(1) for a fine gridpoint. This is the main motivation for using qpOASES with its warmstart functionality.

PR #116 (2020-05-31): [cpp] Update Python Bindings + fix numerical issues.
Fixes #112 (together with #111 and #115)

Changes in this PRs:
- Update the Python bindings.
- Fix numerical issues.
- Fix bug with interpolation discretisation scheme.
- Provide a script that check that C++ and Python implementation returns the same output.

Checklists:
- [x] Update CHANGELOG.md with a single line describing this PR
closed 2020-06-04 by jmirabel