hungpham2511/toppra #122: compute_trajectory doesn't return auxiliary variable trajectory

JokyLee, 2020-06-04:
Hi, I found that function `compute_trajectory` in https://github.com/hungpham2511/toppra/blob/develop/toppra/algorithm/algorithm.py#L145
only returns joint position trajectory. 

But its document indicates it will also return auxiliary variable trajectory. 
I checked the history and found out some editions were made in commit https://github.com/hungpham2511/toppra/commit/f834abb6e8edb965a4ac9a0ad172c802f73e362c#diff-8c1c5f888edbcc3f14137f9f246f401b

Is this function only return one value now?

**Version**
developer branch https://github.com/hungpham2511/toppra/commit/da18170e60fb9bda9a72355eade9388c03860252
and v0.3.0

hungpham2511, 2020-06-05:
Yes. The auxiliary variable trajectory is no longer supported since it's not used in most of our uses of toppra. Actually the docstring is fixed in #117

JokyLee, 2020-06-05:
Awesome! Thank you.

closed 2020-06-06 by hungpham2511