hungpham2511/toppra #132: Duration, Velocity and Acceleration information of waypoints in the trajectory

iamnambiar, 2020-07-23:
Is it possible to find the time duration, velocity and acceleration values of all the waypoints used to generate the trajectory?

iamnambiar, 2020-07-28:
Atleast can we find the duration of the waypoints?  When I checked I can see a variable `sswaypoint` but I think those are the waypoint from `SplineInterpolator` and I cant see any variable to find the durations of the input waypoints. BTW I really liked this library.

hungpham2511, 2020-07-31:
Hi, thanks for using toppra.

Yes you can find the duration, velocity of the waypoints but it requires some work on your part. The reason is that the algorithms only receives the smooth geometric path as input and does not explicitly work with the waypoints.

iamnambiar, 2020-07-31:
Thank you @hungpham2511 .
I have tried a searching method to find the duration of the waypoints.
I used `jnt_traj(ss_waypoint)` to find the position of the interpolated path and using these values I did an interpolation search to find the duration of `waypoints`. But the result are not that accurate. If you got some other ideas, please let me know. It would be really helpful.

hungpham2511, 2020-08-11:
Hi, it's wrong to use `jnt_traj(ss_waypoint)` actually.

You will need to extract manually using the grid_points attributes and the path velocities of the ParametrizationData class https://hungpham2511.github.io/toppra/python_api.html#parameterizationdata

iamnambiar, 2020-08-14:
Thank you. @hungpham2511

closed 2020-08-14 by iamnambiar

chuangwang1991, 2022-02-08:
Hello @hungpham2511 , the lib is a great job, while in my code, the timestamp of output.gridpoints is uniform distribution between 0 and 1, which my jnt_traj.duration is 36 sec, what's wrong with my code?
 `instance = algo.TOPPRA([pc_vel, pc_acc], path,
                        gridpoints=np.linspace(0, 1, 10*(len(pts)-1)+1),
                        solver_wrapper='seidel') # "hotqpoases", 'seidel
instance.compute_parameterization(0, 0)
jnt_traj = instance.compute_trajectory(0, 0)
duration = jnt_traj.duration
print("Found optimal trajectory with duration {:f} sec".format(duration))
output = instance.problem_data
print('output is :', output.gridpoints)
print(output.sd_vec)`

hungpham2511, 2022-02-08:
Can you create a separate issue so that it's easier to keep track? 

Also, what is wrong?