hungpham2511/toppra #134: Contant Time Trajectory with C++ API

JadTawil-theonly, 2020-07-27:
Hello,

Is an analog for the set_desired_duration method available in the c++ implementation?

JadTawil-theonly, 2020-07-27:
Also, it seems like the c++ API requires paths which are composed of segments which are parametrized as polynomials. The python API seems to accept arbitrary waypoints, as opposed to waypoints generated by piecewise polynomials. Is this true? And why so?

hungpham2511, 2020-07-28:
Yes the C++ API as of now only support polynomial path, but you can implement your own path types by inheriting from the abstract interface.

> Is an analog for the set_desired_duration method available in the c++ implementation?

This functionality is not available in the C++ impl.

lbylbylbyl, 2024-04-29:
> Also, it seems like the c++ API requires paths which are composed of segments which are parametrized as polynomials. The python API seems to accept arbitrary waypoints, as opposed to waypoints generated by piecewise polynomials. Is this true? And why so?
hi, do you find or make the c++ use of toppra?can you share me with it?i really need it,thanks

hungpham2511, 2024-04-30:
The python api also constructs a piecewise polynomial internally.