hungpham2511/toppra #135: Inconsistency between Tutorial and source code

farshbafdoustar, 2020-08-05:
**Describe the bug**
running [Basic example of the Tutorial ](https://toppra.readthedocs.io/en/latest/tutorials/tut1_basic_example.html) returns the following error on the latest version of development branch:

compute_trajectory() got an unexpected keyword argument 'return_data'

**To Reproduce**
Steps to reproduce the behavior:
1. gridpoints = np.linspace(0, path.duration, 200)
2. instance = algo.TOPPRA([pc_vel, pc_acc], path, gridpoints=gridpoints, solver_wrapper='seidel')
3. jnt_traj, aux_traj, int_data = instance.compute_trajectory(0, 0, return_data=True)

**Version**
develop

It seems that there is also differences in the return type of jnt_traj. In the latest version of develop branch the type of this value is by default ParametrizeSpline and it has no evaldd() or evald(). 

so the rest of the tutorial is not also compatible:

ts_sample = np.linspace(0, jnt_traj.get_duration(), 100)
qdds_sample = jnt_traj.evaldd(ts_sample)
qds_sample = jnt_traj.evald(ts_sample)

hungpham2511, 2020-08-05:
Thanks for the report. There have been some restructuring of the source code that lead to these inconsistency. You might want to have a look at the scripts inside the folder 'example' for now.

Bests,

Hung

farshbafdoustar, 2020-08-06:
Hi Hung,
Many thanks for your fast reply.
I had already noticed the changed you mentioned in the structure and have checked the example folder to find out if they are adapted to the new one.
But I think the examples are the same as the old versions.
for example in this[ basic example](https://github.com/hungpham2511/toppra/blob/develop/examples/scalar_example.py) 

```
   instance = algo.TOPPRA([pc_vel, pc_acc], path, solver_wrapper='seidel', gridpoints=np.linspace(0, 1.0, 1001))
    jnt_traj = instance.compute_trajectory(0, 0)

    duration = jnt_traj.duration
    print("Found optimal trajectory with duration {:f} sec".format(duration))
    ts = np.linspace(0, duration, 100)
    fig, axs = plt.subplots(3, 1, sharex=True)
    qs = jnt_traj.eval(ts)
    qds = jnt_traj.evald(ts)
    qdds = jnt_traj.evaldd(ts)
```
the same happens for eval, evald, evaldd

My actual question is that how can we get these speed and acceleration values after generating the Spline based trajectory.
Could you please tell me what the latest version (Tag) is that already the examples are adopted to the code structure?

Many thanks,
Behnood

closed 2020-08-07 by hungpham2511

hungpham2511, 2020-08-08:
Hi, you can have a look at the example here: https://hungpham2511.github.io/toppra/auto_examples/plot_kinematics.html#sphx-glr-auto-examples-plot-kinematics-py

Hope this helps.

PR #136 (2020-08-07): [python]Interpolator can use older syntax
Fixes #135 

Changes in this PRs:
- 
- 

Checklists:
- [ ] Update CHANGELOG.md with a single line describing this PR