hungpham2511/toppra #145: Python build instruction fails

felixvd, 2020-10-22:
I followed the [install instructions for the Python API](https://hungpham2511.github.io/toppra/installation.html), but get this error on Ubuntu 18.04:

``` shell
Processing /root/catkin_ws/toppra
Collecting coloredlogs (from toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/5c/2f/12747be360d6dea432e7b5dfae3419132cb008535cfe614af73b9ce2643b/coloredlogs-14.0-py2.py3-none-any.whl (43kB)
    100% |################################| 51kB 4.3MB/s 
Requirement already satisfied: enum34 in /usr/local/lib/python2.7/dist-packages (from toppra==0.3.1)
Requirement already satisfied: matplotlib in /usr/lib/python2.7/dist-packages (from toppra==0.3.1)
Requirement already satisfied: numpy in /usr/local/lib/python2.7/dist-packages (from toppra==0.3.1)
Collecting pathlib2 (from toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/e9/45/9c82d3666af4ef9f221cbb954e1d77ddbb513faf552aea6df5f37f1a4859/pathlib2-2.3.5-py2.py3-none-any.whl
Collecting scipy==0.18.0 (from toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/fc/72/9403ced8a4700b031cc32a12f5711bbb5f7491fb01a2e48030a0dc1acddc/scipy-0.18.0-cp27-cp27mu-manylinux1_x86_64.whl (40.4MB)
    100% |################################| 40.4MB 38kB/s 
Collecting strip_hints (from toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/72/a5/1b8e2990d3f2b24a83d682607a0de5a745ad24de1b2bca33b1561c303a9c/strip-hints-0.1.9.tar.gz
Requirement already satisfied: typing in /usr/local/lib/python2.7/dist-packages (from toppra==0.3.1)
Collecting humanfriendly>=7.1 (from coloredlogs->toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/8e/2d/2f1b0a780b8c948c06c74c8c80e68ac354da52397ba432a1c5ac1923c3af/humanfriendly-8.2-py2.py3-none-any.whl (86kB)
    100% |################################| 92kB 6.7MB/s 
Requirement already satisfied: six in /usr/local/lib/python2.7/dist-packages (from pathlib2->toppra==0.3.1)
Collecting scandir; python_version < "3.5" (from pathlib2->toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/df/f5/9c052db7bd54d0cbf1bc0bb6554362bba1012d03e5888950a4f5c5dadc4e/scandir-1.10.0.tar.gz
Collecting wheel (from strip_hints->toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/a7/00/3df031b3ecd5444d572141321537080b40c1c25e1caa3d86cdd12e5e919c/wheel-0.35.1-py2.py3-none-any.whl
Collecting monotonic; python_version == "2.7" (from humanfriendly>=7.1->coloredlogs->toppra==0.3.1)
  Downloading https://files.pythonhosted.org/packages/ac/aa/063eca6a416f397bd99552c534c6d11d57f58f2e94c14780f3bbf818c4cf/monotonic-1.5-py2.py3-none-any.whl
Building wheels for collected packages: strip-hints, scandir
  Running setup.py bdist_wheel for strip-hints ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nbqmuR/strip-hints/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpPr4KMopip-wheel- --python-tag cp27:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for strip-hints
  Running setup.py clean for strip-hints
  Running setup.py bdist_wheel for scandir ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nbqmuR/scandir/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpSGr_4Hpip-wheel- --python-tag cp27:
  /usr/lib/python2.7/distutils/extension.py:133: UserWarning: Unknown Extension options: 'optional'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for scandir
  Running setup.py clean for scandir
Failed to build strip-hints scandir
Installing collected packages: monotonic, humanfriendly, coloredlogs, scandir, pathlib2, scipy, wheel, strip-hints, toppra
  Running setup.py install for scandir ... done
  Found existing installation: scipy 1.2.3
    Uninstalling scipy-1.2.3:
      Successfully uninstalled scipy-1.2.3
  Running setup.py install for strip-hints ... done
  Running setup.py install for toppra ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-bLYBUt-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yRUefH-record/install-record.txt --single-version-externally-managed --compile:
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help
    
    error: option --single-version-externally-managed not recognized
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-bLYBUt-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yRUefH-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-bLYBUt-build/
```

**Version**
develop branch

hungpham2511, 2020-10-22:
Hi, we no longer support python2.7. Please try again on Python3.6 or higher

closed 2020-10-22 by hungpham2511

felixvd, 2020-10-22:
For anyone googling this, these instructions from [here](https://github.com/larics/topp_ros) sum up the solution for anyone depending on Python 2.7:

> git checkout 8df858b08175d4884b803bf6ab7f459205e54fa2 #checkout to working python 2.7 version of the toppra package
> pip install -r requirements.txt --user
> python setup.py install --user

hungpham2511, 2020-10-22:
Thanks! Yes that will do too.