hungpham2511/toppra #151: Build error installing from pypi

tykurtz, 2020-12-17:
**Describe the bug**
For projects that include `toppra` as a dependency, it is necessary to pre-install `numpy` and `cython`
. 
e.g. `pip install numpy cython; pip install -r requirements.txt`

Some reading material on this issue since it's not unique to `toppra`
* https://pip.pypa.io/en/stable/reference/pip/#build-system-interface
* https://www.python.org/dev/peps/pep-0518/
* https://stackoverflow.com/a/54138355

**To Reproduce**
```sh
rm -rf ~/.cache/pip
mkvirtualenv --python 3.7 toppra-testing
pip install --upgrade pip
pip install toppra
```
which returns
```sh
    ERROR: Command errored out with exit status 1:
     command: /home/tyler/.virtual_envs/toppra-testing/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2vcf69u8/toppra_6dbedb1d31e145f3bddb098516f7c839/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2vcf69u8/toppra_6dbedb1d31e145f3bddb098516f7c839/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-_16s67c0
         cwd: /tmp/pip-install-2vcf69u8/toppra_6dbedb1d31e145f3bddb098516f7c839/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-2vcf69u8/toppra_6dbedb1d31e145f3bddb098516f7c839/setup.py", line 3, in <module>
        from Cython.Distutils import build_ext
    ModuleNotFoundError: No module named 'Cython'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

**Expected behavior**
`pip install -r requirements.txt` works without pre-installing `numpy` and `cython` where requirements.txt includes `toppra`.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
Repeatable on `pypi` as well as locally on `origin/develop`

closed 2020-12-18 by hungpham2511

tykurtz, 2021-03-04:
Hey @hungpham2511 , thanks for merging that fix. 

Could we please get a new release? It would make building virtualenvs which include `toppra` as a dependency much smoother!

hungpham2511, 2021-03-05:
Right, let me try to get to release a new version soon. Thanks for nudging!

PR #152 (2020-12-18): Added pyproject.toml to fix cython bootstrapping
Fixes #151. 

Verified this worked locally by first reproducing the error on `hungpham2511/toppra`, checking out this branch, then running `pip install -e .[dev]` 

Changes in this PRs:
- Adds `pyproject.toml` to install enable installing `toppra` without requiring `cython` and `numpy` pre-installed

Checklists:
- [x] Update HISTORY.md with a single line describing this PR