hungpham2511/toppra #154: install C++ API, which openrobot pkg

oridong, 2020-12-22:
if install C++ API, which openrobot pkg should be installed .
because I see  " /opt/openrobots/lib" is needed, after all  there so many pkd on http://robotpkg.openrobots.org/robotpkg/robots/index.html

jmirabel, 2021-01-04:
Hi,

This package is not provided by robotpkg. It could be added with some little effort but I won't have time to do it myself.

The C++ code can uses QPoases, which can be installed with robotpkg. This is what I used when I contributed to its development. This is probably the reason why you see this.

xuuyann, 2021-01-06:
@oridong Have you successfully installed the C++ version?

xuuyann, 2021-01-06:
@jmirabel I don’t know if your cpp readme document is updated in real time,and it's really unfriendly to beginners.According to your cpp following building tests
# clone
git clone -b develop https://github.com/hungpham2511/toppra

# build
export LD_LIBRARY_PATH=/opt/openrobots/lib:${LD_LIBRARY_PATH}
export CMAKE_PREFIX_PATH=/opt/openrobots
mkdir build && cd build && cmake -DBUILD_WITH_PINOCCHIO=ON -DBUILD_WITH_qpOASES=ON ..
make -j4

# run test
./tests/all_tests
It's unable to install successfully...and I am vexed...
Could you tell me in detail how to install the cpp version？I really appreciate you doing this！Thx！

hungpham2511, 2021-01-06:
Hi @oridong  and @xuuyann, you can refer to the CI pipeline here https://github.com/hungpham2511/toppra/blob/ec723091a717373413d5747d6c1e56d57a646fa1/.circleci/config.yml#L24 for what packages to install.

jmirabel, 2021-01-06:
Installation from source with no meta build tool always requires a bit of involvement. I could write a complete readme for people to use this with robotpkg but this wouldn't suit all users since some will not use robotpkg.

Moreover, when something does not work and you ask for help, you could at least describe what does not work and the error you get. I don't know which step failed. I bet the error message is clear enough for the ones able to demangle it.