hungpham2511/toppra #156: [CPP] Limit on the sd variable

jmirabel, 2021-01-07:
Dear @hungpham2511,

is there a rationale for the hardcoded values
```c++
  x << 0, 100;
  x_next << 0, 1;
```
at line https://github.com/hungpham2511/toppra/blob/ec723091a717373413d5747d6c1e56d57a646fa1/cpp/src/toppra/algorithm.cpp#L40 ?

I couldn't find such a limit in the Python implementation.

hungpham2511, 2021-01-08:
You are right, there is no reason particular reason for that upper limit to be there.

However, we have had problems in the past when x is too large, the optimizer behaves quite badly. In the Python implementation, there is a scaling step to address this problem. See https://github.com/hungpham2511/toppra/blob/644e263269e45be8fc595bae59e959b9570edaa3/toppra/solverwrapper/hot_qpoases_solverwrapper.py#L120

This step should, however, be implemented by the solver wrappers like in the Python implementation because solvers are generally different when it comes to numerical instability.

jmirabel, 2021-01-08:
Thank you. I added a comment in the code.
5099b23f06acb28e822e99e65dbda08c886b69d7

PR #158 (2021-01-14): [cpp] Add Seidel solver.
Fixes #156 and #109.

Checklists:
- [x] Update HISTORY.md with a single line describing this PR
closed 2021-01-08 by jmirabel