hungpham2511/toppra #157: [CPP] Construct piecewise poly from boundary conditions (first derivatives at curve endpoints)

leonardoedgar, 2021-01-08:
Hi. I am trying to find a way to construct a 3rd order piecewise-polynomial / cubic spline by specifying first-derivatives at the endpoints of the curve in cpp.

I found the implementation in **Python** here https://github.com/hungpham2511/toppra/blob/644e263269e45be8fc595bae59e959b9570edaa3/toppra/interpolator.py#L360
With the python implementation, a spline can be constructed by specifying boundary conditions such as the first-derivatives at the endpoints of the curve as documented here https://docs.scipy.org/doc/scipy/reference/generated/scipy.interpolate.CubicSpline.html

However, in **cpp**, all that I can find related to piecewise poly are here https://github.com/hungpham2511/toppra/blob/644e263269e45be8fc595bae59e959b9570edaa3/cpp/src/toppra/geometric_path/piecewise_poly_path.hpp#L23
The spline in cpp can only be constructed from these 2 scenarios:
- Specifying coefficients for each segment of the line and its breakpoints https://github.com/hungpham2511/toppra/blob/644e263269e45be8fc595bae59e959b9570edaa3/cpp/src/toppra/geometric_path/piecewise_poly_path.hpp#L35
- Constructing from the Cubic Hermite Form (specifying the values and its first-order derivatives at each knot) https://github.com/hungpham2511/toppra/blob/644e263269e45be8fc595bae59e959b9570edaa3/cpp/src/toppra/geometric_path/piecewise_poly_path.hpp#L57

In the scenario where the first-order derivatives at each knot are unknown, 
with a general form of a cubic spline in the following form (4n unknowns),
[image]

a cubic spline still can be constructed by specifying the following:
- The values at each knot (2n equations)
[image]

- The first and second derivative of all polynomials are identical in the points where they touch their adjacent polynomial 
(2n - 2 equations)
[image]
[image]
 
- The first derivative at the curve endpoints (2 equations)
f<sub>1</sub>'(x<sub>1</sub>) = a
f<sub>n</sub>'(x<sub>n+1</sub>) = b, where a and b are known values

However, currently, this method is not available in cpp. Are there any suggestions to go about this?

jmirabel, 2021-01-08:
As far as I know, this is not implemented. However, this is easy to implement since this boils down to solving `A x = b` and Eigen provides everything you need for it, you only have to build `A` and `b`. A bit cumbersome but nothing hard.

leonardoedgar, 2021-01-08:
Noted, thanks

hungpham2511, 2021-01-08:
@leonardoedgar If you have an implementation, feel free to open a PR!

leonardoedgar, 2021-01-09:
@hungpham2511 noted