hungpham2511/toppra #172: pip install 'charmap' error

johnwason, 2021-05-04:
I am getting this error on Windows running `pip install toppra` on Windows Python 3.8:

```
Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user\AppData\Local\Temp\pip-install-z85qqebr\toppra_a4f3172db8074f11b27a26e5b412e49c\setup.py", line 13, in <module>
        LONG_DESCRIPTION = file_.read()
      File "C:\Python38\lib\encodings\cp1252.py", line 23, in decode
        return codecs.charmap_decode(input,self.errors,decoding_table)[0]
    UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 1649: character maps to <undefined>
```

johnwason, 2021-05-04:
Commenting out loading `LONG_DESCRIPTION` in setup.py seems to fix the issue.

hungpham2511, 2021-05-06:
Cool. Thanks for reporting the issue. Unfortunately I do not use Windows so I am not sure how should we fix this issue :) Do you have any suggestion?

johnwason, 2021-05-06:
I created a pull request that fixes the character errors. I have a team using some software that depends on this fix, so please upload a fixed wheel so they can use it.

PR #173 (2021-08-08): Remove invalid characters from README.md
Fixes #172

Changes in this PRs:

Remove invalid characters from README.md

Checklists:
- [ ] Update HISTORY.md with a single line describing this PR
closed 2021-08-08 by hungpham2511