hungpham2511/toppra #180: sdotdot not equal to zero at start and end of trajectory?

yakunsjtu, 2021-08-23:
I used the C++ API and found the second derivative of the path parameter is not equal to zero at the start and end of the computed trajectory which is not friendly to the acceleration feedforward control. Is there any chance to enforce the sdotdot to zero?

yakunsjtu, 2021-09-13:
@hungpham2511 , I tried to modify the LinearJointAcceleration.hpp and LinearJointAcceleration.cpp to consider the varying constraints. But the topp failed to compute and returned a total duration of inf. Could it be a problem if the initial acceleration constraint is Zero?

[LinearJointAcceleration.cpp.txt](https://github.com/hungpham2511/toppra/files/7153026/LinearJointAcceleration.cpp.txt)
[LinearJointAcceleration.hpp.txt](https://github.com/hungpham2511/toppra/files/7153027/LinearJointAcceleration.hpp.txt)

jmirabel, 2021-09-14:
It's rather hard to help you with only those files. Can you provide a repeatable example ?