hungpham2511/toppra #196: How to find the time stamp of the waypoints used to generate the trajectory?

chuangwang1991, 2022-02-09:
Hello, in my project, I have 12 waypoints of a 6-dof robot to generate the trajectory.  
 `instance = algo.TOPPRA([pc_vel, pc_acc], path,
                        gridpoints=np.linspace(0, 1, 10*(len(pts)-1)+1),  
                        solver_wrapper='seidel') # "hotqpoases", 'seidel
instance.compute_parameterization(0, 0)
jnt_traj = instance.compute_trajectory(0, 0)
duration = jnt_traj.duration`
In the code, the trajectory duration results is 36.084 sec, I want to find the time stamp of 12 input waypoints used to generate the trajectory.
 `gridpoints=np.linspace(0, 1, 10*(len(pts)-1)+1)`, `len(pts)` is the length of my input waypoints, so the value of `10*(len(pts)-1)+1)` is 10*(12-1)+1=111, so I print the gridpoints , but the results is not the time stamp of the gridpoints, this is the print results:

`Found optimal trajectory with duration 36.083624 sec
output is : [0.         0.00909091 0.01818182 0.02727273 0.03636364 0.04545455
 0.05454545 0.06363636 0.07272727 0.08181818 0.09090909 0.1
 0.10909091 0.11818182 0.12727273 0.13636364 0.14545455 0.15454545
 0.16363636 0.17272727 0.18181818 0.19090909 0.2        0.20909091
 0.21818182 0.22727273 0.23636364 0.24545455 0.25454545 0.26363636
 0.27272727 0.28181818 0.29090909 0.3        0.30909091 0.31818182
 0.32727273 0.33636364 0.34545455 0.35454545 0.36363636 0.37272727
 0.38181818 0.39090909 0.4        0.40909091 0.41818182 0.42727273
 0.43636364 0.44545455 0.45454545 0.46363636 0.47272727 0.48181818
 0.49090909 0.5        0.50909091 0.51818182 0.52727273 0.53636364
 0.54545455 0.55454545 0.56363636 0.57272727 0.58181818 0.59090909
 0.6        0.60909091 0.61818182 0.62727273 0.63636364 0.64545455
 0.65454545 0.66363636 0.67272727 0.68181818 0.69090909 0.7
 0.70909091 0.71818182 0.72727273 0.73636364 0.74545455 0.75454545
 0.76363636 0.77272727 0.78181818 0.79090909 0.8        0.80909091
 0.81818182 0.82727273 0.83636364 0.84545455 0.85454545 0.86363636
 0.87272727 0.88181818 0.89090909 0.9        0.90909091 0.91818182
 0.92727273 0.93636364 0.94545455 0.95454545 0.96363636 0.97272727
 0.98181818 0.99090909 1.        ]`

jmirabel, 2022-02-09:
Can you correctly format the text please ? I reformatted your previous comment but won't do it each time...

hungpham2511, 2022-02-10:
How do you generate the path?

Basically you can extract the gridpoints location of the waypoints from the path, then look for these locations in the final trajectory.

chuangwang1991, 2022-02-10:
Oh, I see,  the gridpoints results could be found by `jnt_traj.waypoints`, I just run my code, that's what I want. 

`[ 0.          1.06357616  1.53501319  1.90142239  2.17173397  2.40051136
  2.62088349  2.83242462  3.03657188  3.24415742  3.46342659  3.68963978
  3.92825418  4.18470423  4.4511162   4.71905627  4.98160503  5.23459676
  5.47238207  5.68679699  5.87105091  6.03727748  6.19741789  6.35368368
  6.50836838  6.66195056  6.81443024  6.96579986  7.11604698  7.26662031
  7.43237699  7.62868835  7.85154645  8.09575959  8.35618709  8.62771482
  8.90524264  9.18367689  9.4579248   9.72288924  9.97346215 10.20266504
 10.41048453 10.61210893 10.82292418 11.04574845 11.27934662 11.52466177
 11.79298339 12.12996284 12.59307962 13.18553514 13.8887111  14.6720125
 15.50542477 16.35917092 17.20355136 18.00883204 18.7451117  19.38207267
 19.88833115 20.24990971 20.54861298 20.83566654 21.11109932 21.37438363
 21.658936   22.00682689 22.41209505 22.85887117 23.33152562 23.81649507
 24.30435515 24.78773322 25.25925353 25.71153134 26.13716471 26.528721
 26.87871194 27.17954136 27.42337748 27.62316925 27.80364521 27.97658293
 28.14148933 28.29765516 28.45031782 28.60859242 28.77816452 28.95830294
 29.14944392 29.37419684 29.65704904 29.98861506 30.35984431 30.76182996
 31.18573536 31.62275992 32.06412061 32.50103961 32.9247337  33.32640201
 33.69720729 34.02824154 34.31045168 34.54862569 34.77403083 35.0075586
 35.248463   35.51224088 36.08362446]`

While the `output = instance.problem_data` seems don't work, because ` output.gridpoints` is a default results.

closed 2022-02-10 by chuangwang1991