hungpham2511/toppra #198: Toppra does not respect acceleration limits (in C++)

jess-moss, 2022-03-25:
**Describe the bug**
I am running toppra on a 10 DoF robot with acceleration and velocity limits in C++, however, toppra is not respecting the acceleration limits. 

To see the bug run the following test (which fails due to lack of respect of the acceleration limits):
https://gist.github.com/jess-moss/448d293853c08b6357e49c475fef6198

**Expected behavior**
When checking the accelerations and velocities along the toppra path, they should be within the limits created in the acceleration and velocity constraints. However, they are not.

**Screenshots**
I plotted the acceleration of one of the failing joints along with its constraints, which is shown below. Notice how the acceleration does not respect the lower bound. 
[image]

In addition, I plotted the position, velocity and acceleration of this joint (see below). Note in this graph, the velocity and position are scaled. I also find it weird how the acceleration jumps a lot for a smooth position and velocity path.
[image]

hungpham2511, 2022-03-26:
Can you check what is the discretizationType of the constraints?

If its `Collocation`, switching it to `Interpolation` should resolve the problem.

jess-moss, 2022-03-28:
The `LinearJointAcceleration` was already set to `Interpolaton`. I set the `LinearJointVelocity` to `Interpolation` just to be sure and it did not resolve the problem. Any other thoughts? I have linked the exact test I am running if you want to play around with it. Thanks!

hungpham2511, 2022-03-29:
> The `LinearJointAcceleration` was already set to `Interpolaton`. I set the `LinearJointVelocity` to `Interpolation` just to be sure and it did not resolve the problem. Any other thoughts? I have linked the exact test I am running if you want to play around with it. Thanks!

Thanks. I will try to have a look at the issue shortly.

hitytqsz, 2022-05-09:
I encounter to  a similar problem like jess-moss:
[image]

hungpham2511, 2022-05-17:
Thanks all for reporting. Hopefully we can find some resource to work on this issue next month. Really sorry for leaving this issue hanging as things have been quite hectic at my work.

mrunaljsarvaiya, 2022-05-31:
@hungpham2511 Out of curiosity so I can better understand the algorithm, do you know of any introductory references to understanding collocation and interpolation discretizations?

hungpham2511, 2022-06-01:
> @hungpham2511 Out of curiosity so I can better understand the algorithm, do you know of any introductory references to understanding the difference between collocation and interpolation discretizations?

I think we have a few references in the paper. Have you tried checking?

These are discretization scheme used in ODE integration algorithms.

hungpham2511, 2022-07-15:
FYI #201 should address this issue more completely.