hungpham2511/toppra #199: New tag/release request

hashb, 2022-06-28:
I am trying to use the recently added get times function. Could you create a new release off the latest master branch

hungpham2511, 2022-07-13:
Hi, sorry the late reply. Sure thing!

hungpham2511, 2022-07-13:
Actually the latest master code has been released as 0.4.1 quite a while ago. Is that not what you look for?

hashb, 2022-07-13:
sorry, my bad, I am using develop branch not master. I am interested in PR https://github.com/hungpham2511/toppra/pull/197

hungpham2511, 2022-07-14:
> sorry, my bad, I am using develop branch not master. I am interested in PR #197

Ah ok. Will do.

hashb, 2022-07-14:
Thank you for the new release. closing this issue now.

closed 2022-07-14 by hashb

hungpham2511, 2022-07-15:
> Thank you for the new release. closing this issue now.

Hi @hashb just curious are you using the Python or C++ version of the library?

hashb, 2022-07-15:
I've been using the python version for a couple of months. I recently switched to the C++ version of the library.