hungpham2511/toppra #206: No module named 'toppra._CythonUtils'

oridong, 2022-08-14:
**Describe the bug**
(base) D:\vsc_pyrob\toppra>python examples/plot_kinematics.py
Traceback (most recent call last):
  File "D:\vsc_pyrob\toppra\examples\plot_kinematics.py", line 10, in <module>
    import toppra as ta
  File "D:\vsc_pyrob\toppra\toppra\__init__.py", line 16, in <module>
    from . import constraint
  File "D:\vsc_pyrob\toppra\toppra\constraint\__init__.py", line 83, in <module>
    from .linear_joint_velocity import (
  File "D:\vsc_pyrob\toppra\toppra\constraint\linear_joint_velocity.py", line 3, in <module>
    from toppra._CythonUtils import (_create_velocity_constraint,
ModuleNotFoundError: No module named 'toppra._CythonUtils'

**To Reproduce**
Steps to reproduce the behavior:
1. 
2. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
Branch, commit, version?

git clone https://github.com/hungpham2511/toppra.git
pip install -r requirements3.txt
pip install -e .

hungpham2511, 2022-08-24:
Hi look like the Cython module weren't built properly. Can you post the output of `pip install -e .`?

oridong, 2022-08-24:
> Hi look like the Cython module weren't built properly. Can you post the output of `pip install -e .`?

win10 anaconda python 3.9

D:\vsc_pyrob\toppra>conda activate

(base) D:\vsc_pyrob\toppra>pip install -e .
Obtaining file:///D:/vsc_pyrob/toppra
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: matplotlib in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from toppra==0.5.1) (3.5.1)
Requirement already satisfied: numpy in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from toppra==0.5.1) (1.22.3)
Requirement already satisfied: scipy>0.18 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from toppra==0.5.1) (1.8.0)
Requirement already satisfied: fonttools>=4.22.0 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (4.33.3)
Requirement already satisfied: pillow>=6.2.0 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (9.1.1)
Requirement already satisfied: pyparsing>=2.2.1 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (3.0.9)
Requirement already satisfied: cycler>=0.10 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (0.11.0)
Requirement already satisfied: python-dateutil>=2.7 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (2.8.2)
Requirement already satisfied: kiwisolver>=1.0.1 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (1.4.2)
Requirement already satisfied: packaging>=20.0 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from matplotlib->toppra==0.5.1) (21.3)
Requirement already satisfied: six>=1.5 in e:\zaprogramsinstalled\miniconda3\lib\site-packages (from python-dateutil>=2.7->matplotlib->toppra==0.5.1) (1.16.0)
Building wheels for collected packages: toppra
  Building editable for toppra (pyproject.toml) ... done
  Created wheel for toppra: filename=toppra-0.5.1-0.editable-cp39-cp39-win_amd64.whl size=4540 sha256=c774b3751b4c98879d8abd2b70b2747fc08a52ed512eb55fe9c278d97dce4a42
  Stored in directory: C:\Users\orili\AppData\Local\Temp\pip-ephem-wheel-cache-ftcv0w9j\wheels\98\b1\73\4507e5eb50b545c99b39e87c4c6af97d8ea96d357d294b825f
Successfully built toppra
Installing collected packages: toppra
  Attempting uninstall: toppra
    Found existing installation: toppra 0.5.1
    Uninstalling toppra-0.5.1:
      Successfully uninstalled toppra-0.5.1
Successfully installed toppra-0.5.1

(base) D:\vsc_pyrob\toppra>

oridong, 2022-08-24:
Sorry, after I just ran "pip install -e .",  the example does complain any more.

closed 2022-08-24 by hungpham2511