hungpham2511/toppra #214: [C++] Geometric path fails when evaluating on its boundaries.

jmirabel, 2022-11-09:
Is there a specific reason why the check below is strict (`<`, `>`) and not loose (`<=`, `>=`) ?

https://github.com/hungpham2511/toppra/blob/8387f7dd9a9063ca665c084fbf53396b55f72ab3/cpp/src/toppra/geometric_path/piecewise_poly_path.cpp#L241

hungpham2511, 2022-11-09:
It's because the path is valid if `pos == m_breakpoints[0]` or `pos == m_breakpoints[N - 1]`.

hungpham2511, 2022-11-09:
As for the issue you mention in the title, is this due to small numerical errors?

DianeBury, 2022-11-10:
This error happens sometimes during the optimization:
```
Error: hpp_idl.hpp.Error(msg='Position 0.0125251 is outside of range [ 0, 0.0125251]')
```
Where the position is equal to one of the bounds of the range. Can this be be due to numerical errors?

jmirabel, 2022-11-10:
@DianeBury can you make sure the above message prints the message with full precision ?

> It's because the path is valid if pos == m_breakpoints[0] or pos == m_breakpoints[N - 1].

I don't understand that. Do you mean invalid ?
The range would thus be `] a, b [` as the code suggests and not `[a, b]` as the message suggests ?

jmirabel, 2022-11-10:
Ok, I misread the check above. I understand now. It must be a numerical error then.