hungpham2511/toppra #216: Usage of C++ API in CMake Project

RAJKgu, 2022-12-04:
Hi,
I am trying to use C++ API in a ROS project because the C++ module is a CMake project we can directly use it ROS project through **find_package(toppra).**

After building, I tried to use it in the ROS package but CMake Could not find a package configuration file provided by "toppra" with any of the following names:
```
    toppraConfig.cmake
    toppra-config.cmake
 ```

Now, for that I have to append the config files path into the **CMAKE_PREFIX_PATH**. But when I look for the directory of config files I am unable to find it. If we looked at the following portion of **CMakeLists.txt** present in the cpp folder

```
set(CONFIG_INSTALL_DIR "lib/cmake/${PROJECT_NAME}")
set(GENERATED_DIR "${CMAKE_CURRENT_BINARY_DIR}/generated")
set(VERSION_CONFIG "${GENERATED_DIR}/${PROJECT_NAME}ConfigVersion.cmake")
set(PROJECT_CONFIG "${GENERATED_DIR}/${PROJECT_NAME}Config.cmake")
write_basic_package_version_file(
    "${VERSION_CONFIG}" VERSION ${PROJECT_VERSION} COMPATIBILITY SameMajorVersion
)

configure_package_config_file(
    "cmake/Config.cmake.in"
    "${PROJECT_CONFIG}"
    INSTALL_DESTINATION "${CONFIG_INSTALL_DIR}"
)

install(FILES "${PROJECT_CONFIG}" "${VERSION_CONFIG}"
    DESTINATION "${CONFIG_INSTALL_DIR}")
install(EXPORT toppra::toppra
  NAMESPACE toppra::
  FILE toppraTargets.cmake
  DESTINATION "${CONFIG_INSTALL_DIR}")
```
Config files directory is set to **lib/cmake/toppra**  what I am deducing is that project cmake files are going to install under `/lib/cmake` right? or its in the build directory of the project because I only found the cmake-config files under the generated folder and cmake-Target file under Export directory but these should ideally be present in one directory toppra under the following path `/lib/cmake`. But I am not able to find any config files under this path, no files are installed there.

jmirabel, 2022-12-04:
May I know how you installed it ?

If you use ROS, you may also use catkin. This is how I use this package personally.

RAJKgu, 2022-12-05:
I follow the instruction mentioned here in the  [Readme](https://github.com/hungpham2511/toppra/tree/develop/cpp#readme). 

Currently, I am using ROS2 humble(Ubuntu 22.04) so I have to use colcon to build the workspace. I haven't tried building C++ API in the ROS workspace but will try to do it. But I thought that if I keep toppra separated from the workspace that should be good. 

Also while building(make), the tests subdirectory failed with the multiple following errors 

```
error: call of overloaded ‘BoundaryCond(<brace-enclosed initializer list>)’ is ambiguous
```

I found this on stack overflow [here](https://stackoverflow.com/questions/14587436/call-of-overloaded-brace-enclosed-initializer-list-is-ambiguous-how-to-deal-w). This may be due to the initialization of `std::vector`.

hungpham2511, 2022-12-05:
Do you have a dockerfile that reproduces the issue? It would make it easier for us to try it out.

FehlerNotFound, 2022-12-08:
Did you try to `sudo make install ` in your local `usr/local/lib`?

RAJKgu, 2022-12-08:
@hungpham2511 Sorry for not getting to you earlier, I didn't have a Dockerfile for reproducing the issue but it doesn't take much time to create one I guess. Currently, I have ubuntu 22.04 on my system with ROS2 Humble distro. As building(CMake) is independent of ROS distro, the reason of test subdirectory failed with the multiple overload errors can be the version of the C++ Compiler used.

@FehlerNotFound I missed this part. Now `toppra`  is installed at `/usr/local/lib` and available to include in CMake projects. Thank you. We need to add this step in [Readme](https://github.com/hungpham2511/toppra/tree/develop/cpp#readme) for Installation. So, that no one will face any issues.

jmirabel, 2022-12-08:
In `sudo make install`, `sudo` is generally a bad idea. What you should do instead is install it an custom install directory and make your environment variable points to it.
This reason is good enough for me not to suggest people to use `sudo` when installing.

RAJKgu, 2022-12-10:
@jmirabel  I agree with you
@hungpham2511 and @jmirabel  I have a question to ask, So I trying to develop a trajectory optimization model which can optimize a recorded set of waypoints. In the pipeline, I am thinking to use STOMP for the optimization of an initial recorded set of waypoints, but for now, I am trying parameterizing the recorded set of waypoints using toppra with velocity(3.14,-3.14) and acceleration(-1,1) bounds. So when I try to run toppra with the recorded points I am getting a parameterized path but viewing it in the plot shows that acceleration is going out of bounds. The parametrized used is "ParametrizeConstAccel". Here is the plot

[image]

So, What am trying to ask is, Is this because of nonuniform recorded waypoints? because for parameterizing we need a smooth geometric path right? This means we can't do parameterization of a raw sequence of waypoints while satisfying the constraints properly like the recorded ones. I am currently exploring new things So don't have much in-depth experience and knowledge of it. But I love to have some guidance that gives me some in-depth insight.

zxfever, 2024-09-29:
> I follow the instruction mentioned here in the [Readme](https://github.com/hungpham2511/toppra/tree/develop/cpp#readme).
>
> Currently, I am using ROS2 humble(Ubuntu 22.04) so I have to use colcon to [...]

hi, I have faced the same error:
```
error: call of overloaded ‘BoundaryCond(<brace-enclosed initializer list>)’ is ambiguous`
```
I don't know if you still remember how you solved it.