hungpham2511/toppra #218: Linear Cartesian move

chen-tianjian, 2022-12-14:
Just wondering if it is possible to formulate the constraints to make the end-effector move in a straight line in Cartesian space (for example, Cartesian velocity direction deviation smaller than a small number)? If so I guess it would be an interesting example.

jmirabel, 2022-12-14:
TOPPRA does not change the path. It only change the time parametrization. So no, TOPPRA cannot constrain the end effector to move on a straight line.

chen-tianjian, 2022-12-14:
Thanks for your reply. Correct me if I am wrong, I am thinking about generating joint profile between 2 way points (that is kind of "re-timing"), such that the end-effector follows a straight line between the 2 way points.

jmirabel, 2022-12-14:
TOPPRA does not modify the geometric path. What you ask for is an IK solver and/or a motion planner. This is out of the scope of this repo. 

You can find other open source software to do this. The choice depends on your use case.

closed 2023-11-10 by chen-tianjian