hungpham2511/toppra #219: N-dof ptah velocity calculation

botoph4, 2022-12-15:
Hey,

when playing with example script `plot_straight_line.py`, I noticed that the joint poses are not directly obtained by 
```
jnt_traj = instance.compute_trajectory(0, 0)
jnt_traj.waypoints
```
rather
```
qs_sample = jnt_traj.eval(ts_sample)  # sampled joint positions
```
Hence `jnt_traj.waypoints` returns `None`. 

So here are some questions I would kindly ask:
1. In which condition will `None` returns instead of an array;
2. Should use `.eval` , `.evald` and `.evaldd` for pose, vel and acc calculation at each point of trajectory;
3. When calculate a n-dof trajectory the return object of ` instance.compute_parameterization( )`, according to doc the path vel has shape of (N, 1) instead of (N ,n).

Cheers

hungpham2511, 2023-02-12:
Very sorry for the late response. Here it is anyway .

> In which condition will None returns instead of an array;

When the trajectory is not constructed from waypoints. The output trajectory of TOPPRA is not so there is no waypoints.

> Should use .eval , .evald and .evaldd for pose, vel and acc calculation at each point of trajectory or the return value of compute_parameterization( );

Yes.

> When calculate a n-dof trajectory the return object of  instance.compute_parameterization( ), according to doc the path vel has shape of (N, 1) instead of (N ,n)

Yes the parametrization should have shape `(N,1)`.

botoph4, 2023-02-15:
Thanks for your kind reply @hungpham2511 ! I will close this issue and it would be great if more infos of calling APIs will be offered in doc.

closed 2023-02-15 by botoph4