hungpham2511/toppra #228: Question on Spline Parametrizer

leonardoedgar, 2023-02-23:
Hi, on this line, may I know why the magic number 5 is used?
https://github.com/hungpham2511/toppra/blob/f8c55e376e40e3f2d5eac0b75d6b82603fe74be6/toppra/parametrizer.py#L181

I understand that delta_t will be set to that value when sdot is very small but is there any specific reason why 5 is chosen? Could the magic number be 2, 3, or 4 as well?

hungpham2511, 2023-02-27:
Hi you shouldn't hit this line in most cases. When you do there is most likely an issue.

Can you provide more information on when do you have this?

leonardoedgar, 2023-02-27:
Hi Hung, actually I had this issue quite frequently when I retimed the final trajectory from ParabolicSmoother. I replicated your pick-and-place scenario in one of your papers. The difference is the suction cup model and the objects' dimensions and weights.

hungpham2511, 2023-03-02:
>  I retimed the final trajectory from ParabolicSmoother

Ah yes, I did have some issue with the ParabolicSmoother before.

Do you have a reproducible example?