hungpham2511/toppra #230: `ConstAccel::evalParams()` computes wrong `k_grid` when `ts[i]` is an element in`m_ts`

TungYiLai, 2023-04-10:
**Describe the bug**
Base on the [description](https://github.com/hungpham2511/toppra/commit/35f8c7d5e245517fb4e0031a3c890934792fdfc1)

https://github.com/hungpham2511/toppra/blob/35f8c7d/cpp/src/toppra/parametrizer/const_accel.cpp#L43-L46
**To Reproduce**
Try this code snippet
```c++
constexpr std::array<double, 7> m_ts = {0, 0.1, 0.2, 0.4, 0.8, 1.6, 3.2};
constexpr double ts = m_ts[2];
const auto ptr = std::lower_bound(m_ts.data() + 1, m_ts.data() + m_ts.size() - 1, ts);
std::cout << "ts: " << ts << ", k_grid: " << (ptr - m_ts.data() - 1) << std::endl; // output: ts: 0.2, k_grid: 1
```

**Expected behavior**
The correct `k_grid` should be 3 instead of 2.

**Version**
[35f8c7d5e245517fb4e0031a3c890934792fdfc1](https://github.com/hungpham2511/toppra/tree/35f8c7d5e245517fb4e0031a3c890934792fdfc1)

sea-bass, 2026-10-31:
@hungpham2511 @inayatku   I actually was running into similar issues and did a sweep over my own examples with release 0.6.10 vs. PRs #302 (closed) and #303 (the one that was actually merged).

I am finding that #302 actually works better:

```
● #302 would have fixed our failures. On the 14 recorded UR5 failure paths (560 runs per version, interpolation only):

  ┌──────────────────────────┬──────────┬─────────────────┬──────────────┬────────────────────────────────┐
  │     TOPP-RA version      │ Failures │ Fixed vs 0.6.10 │ New failures │ Peak acceleration on successes │
  ├──────────────────────────┼──────────┼─────────────────┼──────────────┼────────────────────────────────┤
  │ 0.6.10 (what pixi ships) │ 15       │ –               │ –            │ 1.013×                         │
  ├──────────────────────────┼──────────┼─────────────────┼──────────────┼────────────────────────────────┤
  │ #303 (merged)            │ 25       │ 1               │ 11           │ 1.013×                         │
  ├──────────────────────────┼──────────┼─────────────────┼──────────────┼────────────────────────────────┤
  │ #302 (closed)            │ 0        │ 15              │ 0            │ 1.013×                         │
  └──────────────────────────┴──────────┴─────────────────┴──────────────┴────────────────────────────────┘

```

sea-bass, 2026-10-31:
I think there's a path to take the hybrid, that is the changes in #302 without the Seidel tolerance loosening on failure, and then also keeping the #303 changes. In my tests, that works as well as #302.

hungpham2511, 2026-10-31:
Hi @sea-bass can you provide the failing test cases. Would be happy to have a look at that.

sea-bass, 2026-11-01:
Yep! I just had Claude unload them into a gist -- would this work?

https://gist.github.com/sea-bass/48cacaee82c9f6c0c61faf93b6c34c71

closed 2026-11-04 by sea-bass