hungpham2511/toppra #236: Using JointVelocityConstraint and JointTorqueConstraint can not get result.

1172534699, 2023-06-08:
**Describe the bug**
Using JointVelocityConstraint and JointTorqueConstraint can not get result. I use the Pinocchio library to complete the dynamics calculations
eg: instance = algo.TOPPRA([pc_vel,pc_tau],path,gridpoints=np.linspace(0, 1, 300),solver_wrapper='ecos',)

**To Reproduce**
Steps to reproduce the behavior:
1. My code is as follows.

import toppra as ta
import toppra.constraint as constraint
import toppra.algorithm as algo
import numpy as np
import matplotlib.pyplot as plt
import argparse
import pinocchio
from sys import argv
from os.path import dirname, join, abspath

parser = argparse.ArgumentParser(
    description="An example showcasing the usage of robust constraints."
    "A velocity constraint and a robust acceleration constraint"
    "are considered in this script."
)
parser.add_argument(
    "-N",
    "--N",
    type=int,
    help="Number of segments in the discretization.",
    default=100,
)
parser.add_argument("-v", "--verbose", action="store_true", default=False)
parser.add_argument("-du", "--du", default=1e-3, type=float)
parser.add_argument("-dx", "--dx", default=5e-2, type=float)
parser.add_argument("-dc", "--dc", default=9e-3, type=float)
parser.add_argument("-so", "--solver_wrapper", default="ecos")
parser.add_argument("-i", "--interpolation_scheme", default=1, type=int)
args = parser.parse_args()
if args.verbose:
    ta.setup_logging("DEBUG")
else:
    ta.setup_logging("INFO")

pinocchio_model_dir = join(dirname(dirname(str(abspath(__file__)))), "models")
urdf_filename = pinocchio_model_dir + '/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf' if len(argv)<2 else argv[1]
model    = pinocchio.buildModelFromUrdf(urdf_filename)
print('model name: ' + model.name)
data     = model.createData()

def myInverseDynamic(q,v,a):
    tau = pinocchio.rnea(model,data,q,v,a)
    return tau

# Parameters
N_samples = 4  # 路径点的数量
dof = 6  # 机械臂自由度

way_pts = np.zeros([4,6])
way_pts[0] = [-1.62219,0.0430026,0.51164 ,0.814931 ,0.88985 ,-1.98603]
way_pts[1] = [0.885706, 0.794564, 1.08695, -0.45899, 1.50827, -2.50998]
way_pts[2] = [0.858377, 0.770047, 1.05341, -0.444827, 1.47789, -2.43253]
way_pts[3] = [0, 0, 0, 0, 0.523599, 0]
# way_pts = np.random.randn(N_samples, dof)  # 正态分布 生成 【N_samples, dof】维度的随机数

vlim_ = np.array([4,4,4,6,6,6])
vlim = np.vstack((-vlim_, vlim_)).T  # 对每个自由度设置速度 最小最大限制
alim_ = np.array([6,6,6,6,6,6])
alim = np.vstack((-alim_, alim_)).T
taulim_ = np.array([30,48,5,5,5,5])
taulim = np.vstack((-taulim_, taulim_)).T
fs_coef = list([0] * dof)

path = ta.SplineInterpolator(np.linspace(0, 1, N_samples), way_pts)

pc_vel = constraint.JointVelocityConstraint(vlim)
# pc_acc = constraint.JointAccelerationConstraint(alim, discretization_scheme=constraint.DiscretizationType.Interpolation)
pc_acc = constraint.JointAccelerationConstraint(alim, discretization_scheme=constraint.DiscretizationType.Collocation)
pc_tau = constraint.JointTorqueConstraint(myInverseDynamic, taulim, fs_coef, discretization_scheme=constraint.DiscretizationType.Collocation
                                          )
instance = algo.TOPPRA([pc_vel,pc_tau],path,gridpoints=np.linspace(0, 1, 300),solver_wrapper='ecos',)# 'seidel'  'hotqpoases' 'ecos' 'qpOASES'

X = instance.compute_feasible_sets()
K = instance.compute_controllable_sets(0, 0)

_, sd_vec, _ = instance.compute_parameterization(0, 0)

X = np.sqrt(X)
K = np.sqrt(K)
################################################################################
plt.plot(X[:, 0], c="green", label="Feasible sets")
plt.plot(X[:, 1], c="green")
plt.plot(K[:, 0], "--", c="red", label="Controllable sets")
plt.plot(K[:, 1], "--", c="red")
plt.plot(sd_vec, label="Velocity profile")
plt.legend()
plt.title("Path-position path-velocity plot")
plt.grid()
plt.show()

################################################################################
jnt_traj = instance.compute_trajectory(0, 0)
ts_sample = np.linspace(0, jnt_traj.duration, 100)
qs_sample = jnt_traj(ts_sample)
print(qs_sample)
qds_sample = jnt_traj(ts_sample, 1)
qdds_sample = jnt_traj(ts_sample, 2)
fig, axs = plt.subplots(3, 1, sharex=True)
# plt.grid()
for i in range(path.dof):
    # plot the i-th joint trajectory
    axs[0].plot(ts_sample, qs_sample[:, i], c="C{:d}".format(i))
    axs[1].plot(ts_sample, qds_sample[:, i], c="C{:d}".format(i))
    axs[2].plot(ts_sample, qdds_sample[:, i], c="C{:d}".format(i))
axs[2].set_xlabel("Time (s)")
axs[0].set_ylabel("Position (rad)")
axs[1].set_ylabel("Velocity (rad/s)")
axs[2].set_ylabel("Acceleration (rad/s2)")
plt.show()

and I get the result as follows:
WARNING [reachability_algorithm.py : 292] The initial velocity is not controllable. 0.000000 not in (0.036646, 0.094229)
Traceback (most recent call last):
  File "E:/pinocchio/examples/plot_robust_kinematics.py", line 89, in <module>
    plt.plot(sd_vec, label="Velocity profile")
  File "C:\Users\summer\anaconda3\envs\pin\lib\site-packages\matplotlib\pyplot.py", line 2812, in plot
    return gca().plot(
  File "C:\Users\summer\anaconda3\envs\pin\lib\site-packages\matplotlib\axes\_axes.py", line 1688, in plot
    lines = [*self._get_lines(*args, data=data, **kwargs)]
  File "C:\Users\summer\anaconda3\envs\pin\lib\site-packages\matplotlib\axes\_base.py", line 311, in __call__
    yield from self._plot_args(
  File "C:\Users\summer\anaconda3\envs\pin\lib\site-packages\matplotlib\axes\_base.py", line 465, in _plot_args
    raise ValueError("x, y, and format string must not be None")
ValueError: x, y, and format string must not be None

**Expected behavior**
No matter how small the torque limit, there should be a solution

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
If I set a large torque limit：
[image]

**Version**
I use 'pip install toppra' to install the algorithm.

1172534699, 2023-06-08:
Since you didn't provide an example of dynamics, I'm not sure if my code is correct. I would be very grateful if you would reply to me

1172534699, 2023-06-08:
And I wonder why there is  a big fluctuation in blue line when I set the "gridpoints=np.linspace(0, 1, 300)". But when I set "gridpoints=np.linspace(0, 1, 1000)",the blue line is more smooth.
[image]

hungpham2511, 2023-06-14:
Hi, you can try using `Interpolation` discretization scheme.

hungpham2511, 2023-06-14:
See https://hungpham2511.github.io/toppra/python_api.html?highlight=collocation#toppra.constraint.DiscretizationType.Interpolation