hungpham2511/toppra #260: Compatibility Issue with NumPy 2.x (Import Error)

danielvoconnor, 2025-03-02:
Hi toppra maintainers,

I’m encountering an issue when trying to use toppra with NumPy 2.2.2, and I’d like to ask about potential support for NumPy 2.x or any recommended workarounds.

**Problem**
After installing toppra via pip in a Python 3.12 virtual environment, I get an error when running import toppra. The error suggests a compatibility issue between NumPy 2.2.2 and toppra, which was compiled with NumPy 1.x. Here’s the full traceback:

> A module that was compiled using NumPy 1.x cannot be run in
> NumPy 2.2.2 as it may crash. To support both 1.x and 2.x
> versions of NumPy, modules must be compiled with NumPy 2.0.
> Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.
> 
> If you are a user of the module, the easiest solution will be to
> downgrade to 'numpy<2' or try to upgrade the affected module.
> We expect that some modules will need time to support NumPy 2.
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/Users/doconnor/Dropbox/USF_teaching/Spring2025/MATH373/my_venv/lib/python3.12/site-packages/toppra/__init__.py", line 16, in <module>
>     from . import constraint
>   File "/Users/doconnor/Dropbox/USF_teaching/Spring2025/MATH373/my_venv/lib/python3.12/site-packages/toppra/constraint/__init__.py", line 83, in <module>
>     from .linear_joint_velocity import (
>   File "/Users/doconnor/Dropbox/USF_teaching/Spring2025/MATH373/my_venv/lib/python3.12/site-packages/toppra/constraint/linear_joint_velocity.py", line 3, in <module>
>     from toppra._CythonUtils import (_create_velocity_constraint,
>   File "toppra/_CythonUtils.pyx", line 1, in init toppra._CythonUtils
> ImportError: numpy.core.multiarray failed to import

**Environment**
Python version: 3.12
NumPy version: 2.2.2
toppra version: 0.6.2
OS: macOS

**Steps to Reproduce**
Create a Python 3.12 virtual environment.
Install NumPy 2.2.2 (pip install numpy==2.2.2).
Install toppra (pip install toppra).
Run import toppra in Python.

**Current Workaround**
Downgrading to numpy<2.0 (e.g., pip install "numpy<2.0") resolves the issue, but I’d prefer to use a newer NumPy version if possible.

**Request**

- Is there a plan to update toppra to support NumPy 2.x?
- If not, are there recommended workarounds besides downgrading NumPy (e.g., rebuilding from source)?
- Any guidance would be appreciated!

Thanks for maintaining this great library!

zkingston, 2025-03-06:
I've opened a PR #259 here, take a look and see if that fixes your issue.

JoachimDunkel, 2025-03-15:
I am also having the same issue:

numpy                  2.2.3
toppra                 0.6.0

    from toppra._CythonUtils import (_create_velocity_constraint,
  File "toppra/_CythonUtils.pyx", line 1, in init toppra._CythonUtils
ImportError: numpy.core.multiarray failed to import (auto-generated because you didn't call 'numpy.import_array()' after cimporting numpy; use '<void>numpy._import_array' to disable if you are certain you don't need it).